Ultimaker/Cura #13693: Bad supports, allwais from builplate, 5.2.1 and 5.1.1

UMCURA, 2022-10-30:
### Application Version

521 and 511

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Supports alwais from builplate, select 'normal' and 'everywere'. Nozle 0.2, layers 0.2

### Actual results

Don't work properly

### Expected results

If I select supports 'touchin buildplate' OK, from buildplate, but if I select 'From everywhere' , from objet, is what other software do (ej. Cura 15.04.6) and I can´t solve it in Ultimaker Cura.
Thanks.
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[U2.zip](https://github.com/Ultimaker/Cura/files/9895840/U2.zip)

MariMakes, 2022-10-31:
Get @UMCURA ,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Between Cura 15.04 and Cura 5.2 is at least 6 years of development. 
Probably even more. So you are bound to encounter some differences. 

But I'm having trouble recreating your first screenshot.
If I change my support horizontal expansion, this is the closest I can come.
[image]

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

UMCURA, 2022-10-31:
Sorry, added.
[CFFFP_U2-02_02-2h43.zip](https://github.com/Ultimaker/Cura/files/9901884/CFFFP_U2-02_02-2h43.zip)

MariMakes, 2022-10-31:
Hey! Thanks for sharing the project file. 💪 

I can see the setting that's impacting your quality.
It's Use Towers. If you disable it you get your old behavior back. 
[image]

For some unsollicited advice.
Personally I would increase your Wall Line Count to 3. 
I believe it would get a better result in this model because Cura is trying hard to fill a small space with zigzag infill. 
[image]

Do you still believe it's a bug in Cura? Or would you agree to close the ticket?

UMCURA, 2022-10-31:
Hello, thank you very much, I didn't even have that option visible. I have to investigate more this of the supports, too many options. Thanks again. I close.

closed 2022-10-31 by UMCURA