Ultimaker/Cura #13694: Concentric top and bottom does not work for extruder 2 on UM3extended

paulmalot, 2022-10-30:
### Application Version

5.2.0

### Platform

Windows 10 Pro 64 bit

### Printer

Ultimaker 3 extended

### Reproduction steps

I have a model, round hollow cilinder.  when extruder 1 is selected, and top/bottom finish is as concentric, the preview is correct, and top is concetric.  When i select extruder 2, and keep topbottom as concentric,  the preview show the top/bottom as lines.  Not concentric. see pictures in annex for both situations

[image]
[image]

### Actual results

see added pictures

### Expected results

result in concentric bottom/top on extruder 2

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[UM3E_rol_blanco.zip](https://github.com/Ultimaker/Cura/files/9896004/UM3E_rol_blanco.zip)

MariMakes, 2022-10-31:
Hey @paulmalot,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I think I know what is happening. 
You are looking at the settings for extruder 1, while the model is sliced for extruder 2.
If you check the settings for extruder 2 you will see that the settings indicate lines pattern. 
[image]

It's a mistake I make myself quite regularly. It's something that we are looking to improve to make more clear in the future. 

I believe what you are seeing is undesired but not a bug 
Do you agree that it's not a bug in Cura and we can close the issue?

MariMakes, 2022-11-01:
Don't worry, it's quite a common mistake. I've made it myself as well.
Our designers are looking into ways to make it more clear in the future. 
Since it's not a bug, I'll be closing this issue.

closed 2022-11-01 by MariMakes