Ultimaker/Cura #13697: Align Face to Bed not working

DoubleTElecTech, 2022-10-31:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality Ender 5

### Reproduction steps

Loaded some new models. This is my first model load on a new laptop on version 5.2+ software.  When attempting to orient a model using the align face to bed tool, it will place it at any angle except placing that face along the printer bed. It keeps going at some angle that appears to be about 8 degrees, or it picks what I am guessing is a random triangle in the STL model instead of a face.

### Actual results

See description above. Except if I click on what was the top face of the model that face is rarely even pointed downward.

### Expected results

When a model is selected, it should be able to identify a face or external service and be able to orient that to the build plate for proper orientation when printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2022-10-31:
Hey @DoubleTElecTech,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 
Thank you 🙏

DoubleTElecTech, 2022-11-01:
[Spiral Tap Case R2.zip](https://github.com/Ultimaker/Cura/files/9906390/Spiral.Tap.Case.R2.zip)

See attached for the project file.  All of the items that are at the odd angle, I tried to align to the build plate on a flat surface.  

Apart from this one glitch the program works faster than the previous versions and I am happy with the new version.

MariMakes, 2022-11-01:
Hey @DoubleTElecTech,

Thanks for sharing the project file.
That makes the problem way more clear.

I noticed that if you click the Lay Flat a couple of times you end up with a desirable orientation.
But it's weird that it doesn't happen the first time. 
I ended up getting this orientation after clicking three times: 
[image]
[Spiral Tap Case R2.zip](https://github.com/Ultimaker/Cura/files/9909154/Spiral.Tap.Case.R2.zip)

I'll bring your issue up to the team to get an understanding what is happening and why the first time isn't right, but I hope this helps you to get unstuck.

MariMakes, 2022-11-02:
Hey @DoubleTElecTech,

I checked your issue with the team. Since there is a workaround, we believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I hope you understand. 🙏

regularfry, 2023-10-16:
Is the status on this bug still that it won't be fixed because there's a workaround?  Because 
[this file](https://github.com/Ultimaker/Cura/files/12919502/simple-legs-BackLeg.zip) seems to be immune from the "click Lay Flat until it picks the right orientation" fix.

Layer orientation is critical for this part, I need it to have the large end downwards.

MariMakes, 2023-10-24:
Hey @regularfry,

I took a look, and it seems that indeed the lay flat doesn't work. 
But the button next to it, Select Face to Align, works as intended. 
[image]

I hope this helps 😇

regularfry, 2023-10-24:
That was not my experience.  Select Face To Align would align it to an apparently random face, not the one clicked.  That's why I ended up on this issue in the first place and discovering the recommended workaround.