Ultimaker/Cura #13699: Fail to slide and generate gcode

mhsieh66, 2022-10-31:
### Application Version

5.2.1

### Platform

macOS 13.0 M1Pro

### Printer

Creality CR-10S Pro V2

### Reproduction steps

1. use https://www.thingiverse.com/thing:213946, CubeGears2-3.stl
2. PLA 0.4mm
3. Standard Quality 0.2mm
4. Error while doing a slice, cannot generate gcode

### Actual results

Error while doing a slice, cannot generate gcode

### Expected results

should be able to slice and generate gcode

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CubeGears2-3.zip](https://github.com/Ultimaker/Cura/files/9897862/CubeGears2-3.zip)

GregValiant, 2022-10-31:
Thanks for the report.
I didn't have any trouble slicing this on Windows with 5.2.1 (I believe I printed this myself a couple years ago).
Someone from the Cura team will take a look.  They might like to have your Cura.log file as well.
If you would go to "Help | Show Configuration Folder" and zip the log file and post the zip file here it could be very helpful.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker