Ultimaker/Cura #13700: Z Positon retrogrades in "Vase" mode afte first layer(?)

Ergonomicmike, 2022-10-31:
### Application Version

5.2.0

### Platform

Win 10, 64-bit

### Printer

Ender 3 V2 with JYERSUIEnhanced firmware

### Reproduction steps

I am printing the Lumpy Bumpy Vase. (At 50% scale). I am using "Vase" mode - a.k.a. Spiralize Outer Contour.  (Project files below.)

I have first layer height set for 0.24 mm. The rest at 0.20 mm.

[image]

### Actual results

I made a [video](https://streamable.com/6vzk80). https://streamable.com/6vzk80

The printer prints the first layer, showing 0.24 mm for Z on the printer's Display.

But then it prints the next layer at a LOWER height - lowering to 0.18 in Z!

I believe that the printer really is lowering the Z position because the next layer 'splats' out if I don't increase the Z-offset to compensate.

### Expected results

I expected that the Z position would increment UP to 0.44 mm.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

As this is the first time that I've seen this behavior, I didn't check to see if
    a) if follows Vase mode, and/or
    b) if follows when the user has set a different initial layer height. 

I am not running stock firmware, if that makes a difference.

I had ArcWelder enabled.

Also, I set Line Width to 2x the nozzle diameter, per a famous 3d printer influencer. (BTW, it works!)

[lumpy-bumpy-vase_for_github.zip](https://github.com/Ultimaker/Cura/files/9898189/lumpy-bumpy-vase_for_github.zip)

GregValiant, 2022-10-31:
Thanks for the report.
When I slice your project (Initial Layer Height = 0.24 and Layer Height = 0.50) the Gcode is correct.  Layer:0 goes down at .24mm and Layer:1 goes down at .74mm with all succeeding layers at .50 increments until the start of "Spiralize".
So I am unable to duplicate the problem.  ArcWelder should not have been an issue and it wasn't in my slice.
If you have an ABL then maybe something in it's settings is causing the problem.  How are you printing (SD, Octo, USB)?
Changing the layer height to 0.20 as you describe above, the gcode was still correct.

If you would open the Gcode file in a text editor like Windows Notepad and search for " Z" (space+Z) you will come across all the Z changes.  I've never come across an instance where Cura lowers the Z at any point for printing (Z-hop down is an exception).
I'm inclined to remove the bug label but I'll leave this open if you would like to respond or if you feel that removing the label would be premature.

Ergonomicmike, 2022-10-31:
I should have thought to check the Gcode myself.

I use a form of Manual Mesh in Jyers. I don't have ABL, but the firmware is set for it, and I use the Mesh Leveling protocol to level a 3 x 3 mesh myself.

I print via SD card.

I close this now as a "one of," perhaps something goofy with my firmware when it comes to manual leveling.

Thanks for checking.

closed 2022-10-31 by Ergonomicmike