Ultimaker/Cura #13703: Cura 5.2.1 on macOS Big Sur (11.6.4) crashes with some error logs

mpirio, 2022-11-01:
### Application Version

5.2.1

### Platform

macOS Big Sur 11.6.4

### Printer

Artillery X2

### Reproduction steps

1. Cura was working before
2. Cura crashed (why?)
3. Cura does not start anymore... logs :

```
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /System/Library/Extensions/AppleIntelKBLGraphicsGLDriver.bundle/Contents/MacOS/AppleIntelKBLGraphicsGLDriver: #2: No such file or directory
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /Library/GPUBundles/AMDRadeonX6000GLDriver.bundle/Contents/MacOS/ATIRadeonX6000SCLib.dylib: #2: No such file or directory
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /System/Library/Extensions/AMDRadeonX6000GLDriver.bundle/Contents/MacOS/ATIRadeonX6000SCLib.dylib: #2: No such file or directory
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
```

### Actual results

Cura crash

### Expected results

Cura do not crash...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

```
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /System/Library/Extensions/AppleIntelKBLGraphicsGLDriver.bundle/Contents/MacOS/AppleIntelKBLGraphicsGLDriver: #2: No such file or directory
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /Library/GPUBundles/AMDRadeonX6000GLDriver.bundle/Contents/MacOS/ATIRadeonX6000SCLib.dylib: #2: No such file or directory
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /System/Library/Extensions/AMDRadeonX6000GLDriver.bundle/Contents/MacOS/ATIRadeonX6000SCLib.dylib: #2: No such file or directory
Nov  1 11:48:36 MyComputer Ultimaker-Cura[4626]: getattrlist failed for /System/Library/Frameworks/OpenGL.framework/Resources//GLRendererFloat.bundle/GLRendererFloat: #2: No such file or directory
```

mpirio, 2022-11-01:
My stderr.log if needed :
```
trimesh/curvature.py:12: DeprecationWarning: Please use `coo_matrix` from the `scipy.sparse` namespace, the `scipy.sparse.coo` namespace is deprecated.
  from scipy.sparse.coo import coo_matrix
[35m2022-11-01 12:13:23,485[0m - [32mINFO[0m - [37m[MainThread] UM.Qt.QtApplication.__init__ [99]: Adding QT5 plugin path: /Applications/Ultimaker-Cura.app/Contents/Resources/plugins[0m
[35m2022-11-01 12:13:23,485[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [163]: Initializing Ultimaker Cura[0m
[35m2022-11-01 12:13:23,485[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [164]: App Version 5.2.1[0m
[35m2022-11-01 12:13:23,485[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [165]: Api Version 8.2.0[0m
[35m2022-11-01 12:13:23,485[0m - [36mDEBUG[0m - [37m[MainThread] UM.Application.initialize [166]: Build type None[0m
[35m2022-11-01 12:13:23,485[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/lib/uranium must be a folder.[0m
[35m2022-11-01 12:13:23,485[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/lib64/uranium must be a folder.[0m
[35m2022-11-01 12:13:23,486[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/lib32/uranium must be a folder.[0m
[35m2022-11-01 12:13:23,486[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/MacOS/plugins must be a folder.[0m
[35m2022-11-01 12:13:23,486[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/Resources/uranium/plugins must be a folder.[0m
[35m2022-11-01 12:13:23,486[0m - [33mWARNING[0m - [37m[MainThread] UM.PluginRegistry.addPluginLocation [202]: Plugin location /Applications/Ultimaker-Cura.app/Contents/Resources/cura/plugins must be a folder.[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is /Users/miky/Library/Application Support/cura/5.2[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is /Users/miky/Library/Application Support/cura/5.2[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is /Users/miky/Library/Application Support/cura/5.2/cache[0m
[35m2022-11-01 12:13:23,486[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /Applications/Ultimaker-Cura.app/Contents/MacOS/share/uranium/resources/bundled_packages/uranium.json[0m
[35m2022-11-01 12:13:23,486[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/bundled_packages/cura.json[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile[0m
[35m2022-11-01 12:13:23,486[0m - [36mDEBUG[0m - [37m[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [178]: OpenGL renderer type for this OpenGL version: AMD Radeon Pro 5300M OpenGL Engine[0m
[35m2022-11-01 12:13:23,487[0m - [36mDEBUG[0m - [37m[MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile[0m
[35m2022-11-01 12:13:23,487[0m - [32mINFO[0m - [37m[MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...[0m
[35m2022-11-01 12:13:23,487[0m - [32mINFO[0m - [37m[MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...[0m
[35m2022-11-01 12:13:23,487[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /Applications/Ultimaker-Cura.app/Contents/MacOS/share/uranium/resources/bundled_packages/uranium.json[0m
[35m2022-11-01 12:13:23,487[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from /Applications/Ultimaker-Cura.app/Contents/MacOS/share/cura/resources/bundled_packages/cura.json[0m
[35m2022-11-01 12:13:23,487[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._loadManagementData [201]: Loaded user packages management file from /Users/miky/Library/Application Support/cura/5.2/packages.json[0m
[35m2022-11-01 12:13:23,487[0m - [32mINFO[0m - [37m[MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - [0m
[... cut ...]
[35m2022-11-01 12:13:30,560[0m - [36mDEBUG[0m - [37m[JobQueueWorker [1]] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3pro took 0.17105793952941895 seconds.[0m
[35m2022-11-01 12:13:31,061[0m - [36mDEBUG[0m - [37m[JobQueueWorker [1]] cura.Machines.ContainerTree.run [185]: All MachineNode loading completed[0m
[35m2022-11-01 12:13:39,479[0m - [36mDEBUG[0m - [37m[MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.03315114974975586 seconds[0m
```

mpirio, 2022-11-01:
Cura seems to launch but no window appears

mpirio, 2022-11-01:
Found the issue... In my ~/Library/Application Support/cura/5.2/cura.cfg, I had :
```
window_width = 0
window_height = -28
```

I removed this cfg file and restart Cura to recreate it. all is OK.

closed 2022-11-01 by mpirio

MariMakes, 2022-11-02:
Hey @mpirio,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😅 

I'm super glad that you found a resolution, and that you shared it here. 
If something similar comes up again I'll be able to share your solution 👍 
Have a great day 😊