Ultimaker/Cura #13709: Crash during slicing

flashy63, 2022-11-02:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality CR-10S Pro

### Reproduction steps

1. Opening the joined stl file (see screenshot for initial position)
2. Turning -90° around X-axis
3. Turning -90° around Z-Axis (see screenshot for final position)
4. Setting arborescent supports
5. Launch slicing

### Actual results

Cura start the slicing but, after a little moment (around 40% according to the progressbar), Cura crashed

### Expected results

Have a gcode file !

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9916816/cura.log)
[coude 160mm.zip](https://github.com/Ultimaker/Cura/files/9916826/coude.160mm.zip)
[image]
[image]
[coude 160mm-stl.zip](https://github.com/Ultimaker/Cura/files/9916833/coude.160mm-stl.zip)

GregValiant, 2022-11-03:
Thanks for the report.
It sliced fine for me.  Your Cura log file shows "Backend exited abnormally with return code 3221225477!" which has been coming up on some installations.
This may be a duplicate of some other reports and I'll mark it as such.  The Cura team will take a look.

MariMakes, 2023-02-23:
Hey @flashy63,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors.
Can you download the Beta and retry your slice?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, please let us know.

closed 2023-03-10 by github-actions[bot]