Ultimaker/Cura #13712: When I try to slice my models it freezes and doesn't slice it and comes up with a possible bug error message.

msabourin16, 2022-11-02:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. Attempted to slice it again
2. Quit the software
3. Reopened the software
4. Sliced again

### Actual results

Nothing. It still didn't work.

### Expected results

It should slice the models. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[harry scott hall.zip](https://github.com/Ultimaker/Cura/files/9920144/harry.scott.hall.zip)
[image]

GregValiant, 2022-11-03:
Thanks for the report.
I can duplicate the problem.
I started swapping pieces in and out.  The problem appears to be the model I've marked below with the red arrow.  Rotating that one model 90° seems to fool the bug.
I'll mark this as a duplicate as this behavior has been reported previously.  Someone from the Cura team will take a look at this.  In the meantime, see if that workaround makes a difference for you.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker