Ultimaker/Cura #13713: Second extruder skips prime tower on first few layers

omgitsraven, 2022-11-02:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creator Pro 2

### Reproduction steps

1. Add a model that will require supports
2. Set supports to be printed by the second extruder
3. Add a prime tower and a raft
4. Slice, and preview the slice
5. Watch the first few layers

### Actual results

After the rafts are done printing on extruder 1, extruder 2 begins to print the supports, WITHOUT printing anything on the prime tower first, for several layers.  (Sometimes as little as 1, sometimes as many as 3, depending on the print?)

### Expected results

Extruder 2 should have printed on the prime tower before trying to print any supports -- especially after having sat to one side while extruder 1 printed the raft for several layers, it really needs a moment to get the flow going before trying to print anything worth keeping.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

A demo project: [PrimeTowerBugDemo.zip](https://github.com/Ultimaker/Cura/files/9920427/PrimeTowerBugDemo.zip)

Screenshots showing the problem:

1) The raft is finished:
[image]

2) Two layers of support without the tower changing at all:
[image]

3) The first post-raft extruder 1 layer adds to the tower:
[image]

4) The first extruder 2 layer AFTER that finally starts adding to the tower as well:
[image]

cura.log: [curalog.zip](https://github.com/Ultimaker/Cura/files/9920507/curalog.zip)

For what it's worth, I've also noticed that extruder 2 will skip adding to the prime tower on some later layers as well.

MariMakes, 2022-11-04:
Hey @omgitsraven,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Wow, that looks like a bug 🐛 

https://user-images.githubusercontent.com/40423138/200018334-59446e96-f56a-40c6-8126-5916d673fb1a.mp4

It only seems to happen on rafts, and it's related to the support. We did some changes for rafts in 5.1 so this might be related. 
I'll bring it up to the team to see what we can do to improve it. 
Fingers crossed 🤞

MariMakes, 2022-11-07:
Hey @omgitsraven,

I brought your issue up to the team. 💪 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9860

Thanks for the report! 👍

HFJuli, 2022-12-19:
Hello.
Is there a solution to this problem available now? I need to print every layer on the prime tower to get clean prints.

MariMakes, 2022-12-19:
Hey @HFJuli,

There is no solution available yet.
But it's added to the backlog, so we can pick it up when it reaches the top of the priority list. 

I was about to propose to go back to a previous version of Cura as a workaround. 
But it seems to be present in 5.2, 5.1, 5.0, and 4.13 😖 
I haven't tried earlier versions. 

So the best workaround I can offer is printing a brim instead of a raft.

Vandresc, 2023-03-02:
This is now fixed for the 5.3 stable release, thank you for your feedback!

GoguyT3D, 2023-03-18:
> This is now fixed for the 5.3 stable release, thank you for your feedback!
I am having the same problem on 5.3 with (.1mm layer but not .2mm layer height) bug #14945

omgitsraven, 2023-03-18:
Okay yeah, I just checked too, and, this isn't even fixed in the reproduction project *already* attached to the first post!
[image]

omgitsraven, 2023-03-18:
I also tried creating a new document, with the print settings all reset to default, and only changed the settings necessary to test a prime tower (enabling prime tower, enabling supports on second extruder) and the bug still happens.

MariMakes, 2023-03-20:
Hey @omgitsraven,

I took a look, because I really thought it was resolved. 

The thing that has been fixed in the linked ticket has been the fact that printjob would fail.
In 5.2.1 (left) no PLA is printed in Layer 4 and 5, in 5.3 (right) PLA is printed in layers 4 and 5.
In 5.2.1 layer 6 would be printed in the air because two layers are missing, in 5.3 layer 6 is printed on layers 1-5.
[image]

I agree with you that the actual priming of the pva extruder is missing.

It seems that in the amount of reports we've received over the different sources the focus has been shifted in making sure that the printjob can actually be printed and not in the air. The team has been working on improving the primetower for our 5.4 release (you can read more about it here https://github.com/Ultimaker/CuraEngine/pull/1798).

I'll check for you if a fix for the priming extruder is included there, or if we need to put in some extra effort in actually resolving the priming issue that has been mentioned. 
Apologies for the inconvenience.

MariMakes, 2023-03-24:
Quick update, it's definitely a bug. If the extruder it's used it should be primed. 
Apologies for the miscommunication, thanks for flagging it. 

We've added a ticket to the backlog with the intent of improving this. 
For internal reference CURA-10456

Thanks for the report! 👍

MariMakes, 2023-08-18:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about the prime tower behavior on a raft. 

I asked one of our developers to take a look because we both cannot reproduce this issue in Cura 5.4 anymore. 🤯 
Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into the issue, can you let help us to point out what we should be looking at? 
I'll take another look at what is going wrong. 👍

omgitsraven, 2023-08-18:
You're right, it's fixed!  Thank goodness.  Thank you for catching that!

closed 2023-08-18 by omgitsraven

omgitsraven, 2023-09-05:
For the record:  I'm finding this is still happening in some circumstances, but I'm able to mostly work around it by playing with things like raft thickness until it magically works out....    I won't reopen until I've got a reproduction case, but I thought it was worth noting.