Ultimaker/Cura #13714: Slicing Error

wwian, 2022-11-02:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 S1

### Reproduction steps

Set tree support to 5°, sliced. Slice failed.

### Actual results

Slicing was not completed.

### Expected results

Slicing to be completed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[8a169f89-1885-4a86-be56-a6ae63536e6b.zip](https://github.com/Ultimaker/Cura/files/9921590/8a169f89-1885-4a86-be56-a6ae63536e6b.zip)

GregValiant, 2022-11-03:
Thanks for the report.
I was able to slice the model with Normal supports but when switching to tree supports Cura hung and brought up the _Report a Bug_ dialog so I was able to duplicate the problem.

I rotated the model 90° on the Cura build plate and that fooled the bug and the slice completed with tree supports.
This may be a duplicate of several other reports regarding a slicing failure that could be solved by moving or rotating the model and I'll mark it as such.  (FYI the print with Normal supports was 3 hours shorter.)
[image]

wwian, 2022-11-03:
Thanks so much for the quick reply and suggestions!

I'll give the rotation a try and see if that works. I'll also try the
normal supports.

Cheers,

Ian!