Ultimaker/Cura #13715: Cura start up re-inserts the gcode for Anycubic Vyper profile

tkocou, 2022-11-02:
### Application Version

5.2.1

### Platform

Linux Mint

### Printer

Anycubic Vyper

### Reproduction steps

After starting Cura for the day, I checked the Printer's **Machine Settings**. In past Cura versions, the Printer **Machine Settings** remained blank after deleting the gcode those areas. With Cura 5.2.1, **every time** I start up Cura, the **Machine Settings** for my Vyper printer shows the unwanted gcode. 

### Actual results

This unwanted behavior forces me to edit the printer profile before attempting any printing. Not good!!

### Expected results

When a printer profile has the gcode areas deleted (Start G-code & End G-code), those areas should remain deleted.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9921986/cura.log)

tkocou, 2022-11-02:
The behavior started with the update from 5.1.1 to 5.2.1

GregValiant, 2022-11-02:
Thanks for the report.
There must be something "different" about the AnyCubic printers.  I can reproduce the problem with them but not with my Ender (which is the only other one I checked).
The default StartUp gcode seems to be inserted but the Ending gcode always comes in as a blank line and then "machine_startup_gcode =" which is certainly wrong.

I'm going to leave the bug label on so the Cura team will take a look but I have a feeling that this is an AnyCubic problem rather than a Cura problem.  I can't see why Ultimaker would care about AnyCubic Start and end Gcodes.

I did find a workaround.  If you put a comment into the boxes like...
; No StartUp
and 
; No Ending

That will fool the outside process (or maybe bug) that keeps re-inserting things.

GregValiant, 2022-11-03:
After clearing the StartUp Gcode and Ending Gcode from several printers, quitting Cura, and restarting Cura I get this in the Machine Settings Dialog:
[image]

The StartUp gcode came back into the empty text box and the Ending Gcode is obviously wrong.

teamRibcage, 2022-11-03:
I can confirm that this is what occurs with my installation as well ( Cura 5.2.1 // Ender-3 S1 // Windows 10 )
Additionally, as noted above, adding any bit of code such as:
```
; No Added Cura Code
```
prevents the incorrect code from being re-inserted. Hope that helps! 😊

tkocou, 2022-11-04:
Another work-around:
I purged version 5.2.1 and started version 4.8.0. In 4.8.0, I created a custom FFM printer with all the correct settings for the Anycubic Vyper, but without the Gcode. After shutting down 4.8.0, I installed 5.2.1 and let it import the printers, etc. Now the custom Anycubic Vyper definition stays blank when I start Cura version 5.2.1