Ultimaker/Cura #13719: Individual Features (Not a good enough title, please read.)

Scharfschutzen, 2022-11-02:
### Is your feature request related to a problem?

So I was wondering if Cura could have the ability to apply features to an edge? For example, I want to apply a brim or raft to a specific body on a model (to prevent it from shrinking/lifting from bed), but not the whole model. If I could apply a brim to just that portion of the print, but leave the rest of the print as a brim (since it has no shifting), that would be super helpful to save material and post-print cleanup.

Solidworks behaves in a similar fashion, allowing you to select edges for fillets and chamfers, etc.

### Describe the solution you'd like

I honestly don't have slicer alternatives, but as mentioned, Solidworks is capable of said feature.

### Describe alternatives you've considered

As said before, Solidworks can create this feature in the modeling phase, but not within the slicing phase.

### Affected users and/or printers

Users wanting to apply features to only specific parts of a model and not the whole thing.

GregValiant, 2022-11-03:
Cura has only rudimentary ability to alter models.  SolidWorks is a CAD package and Cura is simply a slicer.

In the Cura MarketPlace is a plugin for "Tab Anti-Warping".  You can add the tabs to a model.  You can print them as "drop down support" (the default) or as a regular model and they can be scaled like any other mesh.  They need to be post processed to cut them off but any helpers would.
Here I've added three round ones and then stretched the middle one in the X.
The Cura team will look at your request but in the meantime maybe something like the tabs would be helpful.
[image]

MariMakes, 2022-11-08:
Hey @Scharfschutzen,

Welcome to the Ultimaker Cura Github 🚀 

To me, it doesn't seem like an issue in Ultimaker Cura, since we have a solution in place, with this plugin. 
You can get it here: https://github.com/5axes/TabPlus and on the marketplace. 

Would you still like to keep this issue open or do you agree that we can close it?

closed 2022-11-23 by github-actions[bot]