Ultimaker/Cura #13720: Unable to slice at all.

SynPrint, 2022-11-03:
### Application Version

5.2.1

### Platform

windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

Load file - any file
standard quality
change nothing. 
hit slice. 

### Actual results

I get the unable to slice error

"SLICING FAILED - SLICING FAILED WITH AN UNEXPECTED ERROR. PLEASE CONSIDER REPORTING A BUG ON OUR ISSUE TRACKER." 

### Expected results

for it to slice

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9925156/cura.log)
https://www.dropbox.com/s/ugine4blktrm1no/CE3PRO_E-TRP-Castle-Floor-v4.0.3mf?dl=0

GregValiant, 2022-11-03:
Thanks for the report.
I didn't have any trouble slicing your model.  There has been a bug reported (several times) that can be fooled by rotating the model on the build plate.  See if rotating the model works for you and let us know.  If it does I will mark this as a duplicate.

SynPrint, 2022-11-03:
> v

Tried rotating in multiple directions an I get the same error every time. 

It should be noted that this only started happening when I upgraded to 5.2.1 but not will not work for any version. I've uninstalled all cura versions and reinstalled a new clean slate 5.2.1 and I'm still getting this issues. Slice with Prusa and it has no issues.

GregValiant, 2022-11-03:
That makes this sound like it's something different.
In the log file is "Backend exited abnormally with return code 3221225477!".  That code has come up previously in 5.1 and 5.2.  The Cura team will take a look.

GregValiant, 2022-11-12:
@MariMakes may know.

rburema, 2022-11-15:
I can confirm that slicing fails for me as well in 5.2.1 as well as in the current build from source. I'll bring it up with the rest.

In the meantime, I've found that, for me at least, rotating the model by 90 degrees counter-clockwise, does allow it to slice. I've attached the project here. (You can attach files directly, as long as its one of a few pre-approved types, fortunately, zip is one of those.)

[CE3PRO_E-TRP-Castle-Floor-v4.0__rotated90.3mf.zip](https://github.com/Ultimaker/Cura/files/10011190/CE3PRO_E-TRP-Castle-Floor-v4.0__rotated90.3mf.zip)

SynPrint, 2022-11-20:
> 

Unfortunately the file I used was just a reference, Mine is doing this no matter what file I try to slice. I've tried the rotation trick but that doesn't work for me at all in any dimensions.

rburema, 2022-12-22:
If it's really any file:

Can you try running the engine (CuraEngine.exe) from the terminal, just to check if it hasn't been corrupted?

(Without any arguments.)

MariMakes, 2023-02-23:
Hey @SynPrint,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, can you let us know?

rburema, 2023-02-23:
For those in the know, to give a bit more context, `3221225477` is the decimal form of return code `0xC0000005`, which is Microsoft's way of saying 'segmentation-fault'. Which means the engine (backend) is trying to access a bit of memory that 'doesn't exist' or the program shouldn't have access to, which is a common error when working with C++ (unfortunately).

closed 2023-03-10 by github-actions[bot]