Ultimaker/Cura #13722: Unable to slice the model

wxydejoy, 2022-11-03:
### Application Version

5.2.1

### Platform

windows 10

### Printer

Ultimaker s3

### Reproduction steps

1. drop the STL file into software
2. slice
3. error

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[model.zip](https://github.com/Ultimaker/Cura/files/9927228/model.zip)

GregValiant, 2022-11-03:
Thanks for the report.
I didn't have any trouble slicing this model in the orientation that the STL file opened with or with the model rotated.

In Cura, use the "Help | Show Configuration Folder" command to find the Cura.log file.  Zip that file and post it here.  The Cura developers will need it to find out what happened.

wxydejoy, 2022-11-05:
I tried to reproduce the problem, but it was not successful, so I have to give you all the logs. The file is a little large. Sorry
[cura.zip](https://github.com/Ultimaker/Cura/files/9942828/cura.zip)

GregValiant, 2022-11-05:
I found what I thought might be there.  I don't know what it means, but when this behavior has been reported previously this line has been in the log and it is in your log file:

"Backend exited abnormally with return code 3221225477!"

So I would call this a duplicate of some previous reports.

MariMakes, 2023-02-23:
Hey @wxydejoy,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅 
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors. 
Can you download the Beta and retry your slice? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, can you let us know?

wxydejoy, 2023-02-25:
It works

closed 2023-02-25 by wxydejoy