Ultimaker/Cura #13725: Cura 5.2.1 not starting

Anfeket, 2022-11-03:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

None

### Reproduction steps

1. Start Cura

### Actual results

Nothing happens except loading cursor

### Expected results

Cura starts

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No log file, except 5.2 and plugin folder being created in %appdata%\cura.
Program shows up in task manager but disappears shortly after.
I would like to try installing some redistributables but didn't find any.

MariMakes, 2022-11-04:
Hey @Anfeket ,

Thanks for your report 👍
😖 Sorry to hear you are stuck. 

I've seen a number of reports like this. 
I have a couple of questions to narrow it down.
1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
2. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
3. Do you have a Ryzen Graphics card? [Can you try this workaround](https://support.ultimaker.com/hc/en-us/articles/360018633500-Ultimaker-Cura-won-t-run-on-my-Ryzen-Graphics-card)? 

If your solution is not on this list let me know and we can try something else 💪

Anfeket, 2022-11-04:
I do not have multiple monitors, not sure what gpu OSD is but I didn't tweak my gpu really and I don't have a Ryzen gpu. I have tried running the executable through command prompt with no output unfortunately. I think it would be quite helpful to print some debug info to stdout if possible.

Anfeket, 2022-11-05:
I should probably add that the last 4.x release still works

Anfeket, 2022-11-07:
I only have Windows Defender and it didn't report anything related but after looking I was unable to find Ultimaker-cura.ico anywhere, I also excluded the entire folder and reinstalled but still no luck.

Anfeket, 2022-11-10:
Maybe [DxDiag.txt](https://github.com/Ultimaker/Cura/files/9983438/DxDiag.txt) output might help.