Ultimaker/Cura #13727: Gcode filament retractions cause  small or no skirt or brim on next print when printing small models.

WardXmodem, 2022-11-03:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Prusa I3

### Reproduction steps

print any model - gcode for Prusa I3 or Anet A8 retracts more at end than it feeds at beginning. 
I would want a full skirt (say, 3 lines) or brim not having just a fraction of one line print or none at all. 

### Actual results

This "annoyance" - I'm now calling a bug -- is that the supplied gcode retracts too much at the end, and doesn't make up for it in the starting gcode.  I have to remember to manually push some filament before printing, then remove it from the nozzle -- if I can -- right after the feed after the homing and feeding some filament as it starts to print. 

I switched from Prusa I3 to my actual printer, Anet A8, and the same problem exists. 

The ending code: 
`G92 E0
G1 E-10 F2000`

so it retracts 10 

Then the next print's beginning code: 
```
G92 E0
G92 E0
G1 F1500 E-6.5

;TYPE:SKIRTG1 F1500 E0
G1 X88.507 Y104.3 E0.02458  ;<= start of printing skirt
```
This is an E-10 and an E-6.5, or a total of E-16.5, as of the time it starts printing the skirt. 
Sure enough looking at the G-code, the last extrude of a 3-line skirt is 
`G1 X88.615 Y105.234 E17.66847`
it FINALLY gets to extruding > 16.5 just as the skirt ends. 

### Expected results

I ask for 3 lines of skirt, I get from none to less than one

The model's skirt is small - only about 10 x 30 mm, a larger model with 3 lines of skirt would obviously extrude a LOT more. 

### WORKAROUND: 
add more lines to skirt.   BUT WHY DO I HAVE TO?  WHY SO MUCH EXTRACTION KNOWING IT WILL CAUSE ERRORS WHEN STARTING TO PRINT? 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Thank you, Thank you, Thank you for all you do?  

I've had this problem since day 1, probably Cura 3.x.x - and just got tired of the problem - mostly because I'm printing such a small model 

### Frustrating bug in submitting this report
even though the form detected a Windows 10 platform and pre-fills that in, you can't "submit new issue" unless you figure out you have to OVERTYPE the Windows 10 ! ! ! ! ! ! !

GregValiant, 2022-11-04:
Thanks for the report.
_"...retracts too much at the end, and doesn't make up for it in the starting gcode."_
The StartUp and Ending Gcode are defined in your Printer Definition file that was submitted to Ultimaker for inclusion in Cura.  Ultimaker has no idea how much retraction/prime is required by a 3rd party printer.  There is no bug here, just a big annoyance.

At any rate, I don't find those lines in the stock Ending Gcode for any Prusa definition in Cura.  It would appear that you have a "custom" ending gcode that you don't like.

The line you posted above:  "G92 E0 G1 E-10 F2000" is an error.  It should be.
G92 E0
G1 E-10 F2000

A fix would be to go into Manage Printers / Machine Settings and in the Ending Gcode box change this line:
G1 E-10 F2000
to this
G1 E-{retraction_amount} F2000
(note the negative sign before the opening curly bracket.)
Instead of a "hard coded" 10mm retraction, you would get whatever your Cura setting "Retraction Distance" is set to.

I'll remove the bug label as this is a problem with your Ending Gcode.  Let us know if you feel that is premature.

WardXmodem, 2022-11-07:
Thank you thank you!!!!   
1) I don't know how the G1 command got wrapped up with the previous, sorry 'bout that. 
2) Ah, vendor supplied GCODE, That's a wonderful answer
3) G1 E-{retraction_amount} F2000  - is a great solution!!!!  Implemented.

closed 2022-11-07 by WardXmodem