Ultimaker/Cura #13728: Stack Overflow on CuraContainerRegistry.loadAllMetadata

Gentoli, 2022-11-04:
### Application Version

5.2.0+

### Platform

Windows 11 Insider (25231.1000)

### Printer

N/A

### Reproduction steps

1. Cura 5.1.1 runs normally
1. Install 5.2.0 or 5.2.1
1. failed to start with stack overflow

### Actual results

Stack Overflow:
```
Traceback (most recent call last):
  File "UM\Settings\ContainerRegistry.py", line 469, in loadAllMetadata
    db_last_modified_time = self._getProfileModificationTime(container_id, cursor)
  File "UM\Settings\ContainerRegistry.py", line 415, in _getProfileModificationTime
    db_cursor.execute("select id, last_modified from containers where id = ?", (container_id, ))
sqlite3.OperationalError: no such table: containers

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "cura_app.py", line 239, in <module>
  File "cura\CuraApplication.py", line 824, in run
    super().run()
  File "UM\Qt\QtApplication.py", line 381, in run
    super().run()
  File "UM\Application.py", line 268, in run
    self.startSplashWindowPhase()
  File "cura\CuraApplication.py", line 545, in startSplashWindowPhase
    self._container_registry.loadAllMetadata()
  File "cura\Settings\CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM\Settings\ContainerRegistry.py", line 472, in loadAllMetadata
    self._recreateCorruptDataBase(cursor)
  File "UM\Settings\ContainerRegistry.py", line 412, in _recreateCorruptDataBase
    self.loadAllMetadata()
  File "cura\Settings\CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM\Settings\ContainerRegistry.py", line 472, in loadAllMetadata
    self._recreateCorruptDataBase(cursor)
  File "UM\Settings\ContainerRegistry.py", line 412, in _recreateCorruptDataBase
    self.loadAllMetadata()
  File "cura\Settings\CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM\Settings\ContainerRegistry.py", line 472, in loadAllMetadata
    self._recreateCorruptDataBase(cursor)
...
```

Also seeing a bunch of:
```
2022-11-03 22:22:48,191 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket was unable to bind to port, increasing port number to 49768
2022-11-03 22:22:48,192 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [235]: Previous socket existed. Closing that first.
2022-11-03 22:22:48,196 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
2022-11-03 22:22:48,197 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [235]: Previous socket existed. Closing that first.
2022-11-03 22:22:48,202 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket was unable to bind to port, increasing port number to 49769
2022-11-03 22:22:48,203 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [235]: Previous socket existed. Closing that first.
2022-11-03 22:22:48,205 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
2022-11-03 22:22:48,206 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [235]: Previous socket existed. Closing that first.
```

### Expected results

Cura start or report actionable error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9934456/cura.log)

MariMakes, 2022-11-04:
Hey @Gentoli,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I asked around a bit, because initially I was a bit lost on this one. 
Can you try deleting the C:\Users\<username>\AppData\Local\cura\<version>\cache folder ?

If you start Cura again then that should properly rebuild the DB and start without the Stack Overflow.  If that doesn't work it might be related to your permissions in that folder. 

Can you let us know if this works for you? Thank you 😊

Gentoli, 2022-11-04:
@MariMakes Thanks, that did the trick.

The cache directory contains only a zero sized containers.db file. IIRC my drive was very low on space when installing the update so it probably failed to write that file.

I think this should be addressed by either
- clean config option in crash reporter clears the cache folder (if it is cache only)
- Cura logs where the corrupted file is, so the user can decide what to do

Gentoli, 2022-11-04:
@MariMakes Thanks, that did the trick.

The cache directory contains only a zero sized containers.db file. IIRC my drive was very low on space when installing the update so it probably failed to write that file.

I think from a user perspective, this should be addressed by either:
- clean config option in crash reporter clears the cache folder (if it is cache only)
- Cura logs where the corrupted file is, so the user can decide what to do

MariMakes, 2022-11-07:
Hey @Gentoli,

I'm happy to hear you were able to resolve it 😊 
I'll bring up your suggestion to the team and add an update if we end up addressing it. 

But since your bug is resolved. I'll be closing this issue. 
Happy printing 👍

closed 2022-11-07 by MariMakes

MariMakes, 2022-11-09:
I brought your suggestion up to the team. 
We've added a ticket to the backlog with the intent to help users running into this issue. 

Thanks again for the report! 👍