Ultimaker/Cura #13730: Memory Leak

kpbopp, 2022-11-04:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Leave Cura open for an extended period of time

### Actual results

I'll leave cura open over night and come in to work the next morning to my computer fans going crazy because it's using 2GB of memory.

### Expected results

I would expect Cura to be using its normal amount of memory.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9938120/cura.log)
This has been going on since we got these printers and started using Cura almost 2 years ago.

MariMakes, 2022-11-04:
Hey @kpbopp,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13730
It's a long thread to read but their issue seems to have resolved by upgrading to Cura 5.2.1 and not adding their printer via ipadress. 

I can see in your logs that: It has been 10749.094150066376 seconds since the last response for outputdevice manual [ipadress]
That usually means that your printer or your laptop didn't have access to the internet for over 3 hours and ended in a time-out.
But this should not create a memory leak. 

I'm curious, why is it important for you to have your Ultimaker Cura open overnight? Are you monitoring a printjob or did you start something you still want to finish? 

Can you try adding your printer to Cura in an alernate way? Perhaps even upgrade to Cura 5.2.1?
I'll ask the team to take another look at your issue and the linked  https://github.com/Ultimaker/Cura/issues/13730

fieldOfView, 2022-11-04:
Euhm, this _is_ #13730.

kpbopp, 2023-01-11:
Sorry for not replying a while back. I added the 2 printers through digital factory, without updating Cura, and that seemed to fix the memory leak. The annoying thing now though is that I can't view the camera feed in Cura anymore, it has to be through the digital factor web page. Oh well I suppose, not a huge deal. Will try updating Cura at some point to see if that helps.

MariMakes, 2023-09-11:
🤦‍♀️ I see I linked the wrong ticket, I meant to link https://community.ultimaker.com/topic/44168-memory-leak-cura-54/

@kpbopp, you mentioned that your leak is resolved, would you still like to keep this issue open or do you agree that we can close it?  

Regarding your camera feed, if you are on the same network you can add your printer with IP to see the old camera feed. The Digital Factory feed is accessible from anywhere but only updates every 30 seconds.

closed 2023-09-25 by github-actions[bot]