Ultimaker/Cura #13731: Slicing Failed bug report

BillyCroan, 2022-11-04:
### Application Version

Ultimaker-Cura-5.2.1-linux-modern.AppImage

### Platform

Fedora 36 GNU+Linux

### Printer

Ultimaker S3

### Reproduction steps

Added stl file.
failed to slice.

### Actual results

"Slicing failed with unexpected error please file a bug"

### Expected results

slice normally

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

files at http://bcroan.com/cura/
77dcf9ede6b20201d89c39a92d5ff00c  Blender Gasket.stl
0c356433302fc8c6fb5fcdb0b63c4351  Blender Gasket Slice Error.3mf
102db46162995cc128049ea49b4e8e83  cura-2022-11-04.log

BillyCroan, 2022-11-04:
I observed the same error on 5.2.1 on Windows 10 Enterprise LTSC v 1809, but it was intermittent and I was able to print the stl with others on the same plate.
I've added cura-windows.log to http://bcroan.com/cura/ which is the log from the windows session.

MariMakes, 2022-11-04:
Hey @BillyCroan,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I can recreate your error on my side. So I'll bring it up to the team to take a look at it. 

I noticed two things.
1. Your mesh doesn't seem to be watertight. This results in some unretracted travels halfway through your model. 
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools. Or exporting it with different resolution from blender? 
[image]
2. Your model does slice if you select a different layerheight or resize it to 100.3%. That makes me think it might be a rounding issue. I hope this at least gets you unstuck. 
[image]