Ultimaker/Cura #13732: USBPrinting on Macos Ventura with M1 fails to detect connected printer. Printer does respond through serial Arduino console

Roos-AID, 2022-11-04:
### Application Version

5.2.1

### Platform

Macos Ventura M1

### Printer

Anet A6

### Reproduction steps

On Macbook with Intel and Monterey Cura 5.2.1 finds the Anet A6 printer.   Just acquired a new Macbook with M1Pro and Macos Ventura.  
Clean install of Cura 5.2.1 
Select Anet A6 printer.
Printer is not detected.
In the log USBPrinting 1.0.2 is loaded but does not seems to be intialized in the log. 

When I open serial console from Arduino,  then the printer is seen on USB port and responds to commands. 
ls /dev/cu* returns  /dev/cu.usbserial-210
Serial port in Arduino is /dev/cu.usbserial-210 as well 

### Actual results

Printer is not detected

### Expected results

Expected printer interface 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

In the log I do see the USBPrinting 1.0.2 to be loaded.
I do not see any reference to USBPrinterOutputDevice method.   On Macos Intel I see the OnExitCallbackManager calls. This does not show on this system.
It looks as if the USB port is skipped as not being a printer device ? 

It might be due to new security mechanism in Macos Ventura,  I cannot reproduce this on my old machine unfortunatly. 
[cura.log](https://github.com/Ultimaker/Cura/files/9939004/cura.log)

MariMakes, 2022-11-04:
Hey @Roos-AID,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Can you double-check for me if you have the USB printing module enabled?
If you go to the marketplace and click on the gear you can see your bundled plugins. 

[image]

If that's not it, I can have someone look at your logs.

Roos-AID, 2022-11-04:
I can confirm that it is enabled.  I have also tried to disable, restart, enable, restart. Does not help. 
In the log you can see that it is loaded.

MariMakes, 2022-11-04:
Hey @Roos-AID,

Thanks for the update. 
I'll have one of our Mac Developers take look 👍

Roos-AID, 2022-11-10:
Also tested with Macos Ventura 13.0.1 , still no printer  detected.

casperlamboo, 2022-12-14:
Hi @Roos-AID,

Thank you for the bug report. We have looked into the logs but did not find any indication what could have caused the issue. We will close the issue for now but we'll keep an eye open for similar reports.

Hope you understand our decision. Feel free to re-open the issue if you have more information and/or have reproductions steps for the issue.

Thanks,
Casper

neoneddy, 2022-12-26:
Happening for me as well.  Confirmed USB driver enabled.  I can also confirm MacOS Ventura on M1 macbook is showing a USB Serial device in /dev/

Roos-AID, 2022-12-27:
> Happening for me as well. Confirmed USB driver enabled. I can also confirm MacOS Ventura on M1 macbook is showing a USB Serial device in /dev/

Question for Neoneddy:  Do you have Parallels installed ?  

For the developer of Cura :
The problem is persistent. 
Reproduction steps: 
1. Get Macbook with M1 processor
2. Clean install of Macos Ventura
3. Install  Cura 5.2.1
4. Select Anet A6 printer and connect via USB
5. Printer is not detected by Cura
6. When opening serial console in Arduino or similar and selecting the USB port, the printer is accessible. 

It seems that the /dev/ usb devices is not recognised as a printer and hence the  initialisation does not take place in Cura.

nickvellios, 2023-01-02:
I am having this same issue.  2020 M1 MacBook Air, macOS Ventura, Cura 5.2.1, Anycubic Kobra Max with firmware v2.8.7.  I have tried everything I can think of.  The device is recognized:
```bash
➜  ~ ls /dev/*usb*     
/dev/cu.usbserial-110  /dev/tty.usbserial-110
```

MariMakes, 2023-01-03:
Hey All,

I found a workaround on the Ultimaker Community Forum for recognizing printers on Mac devices. 
Hope this helps someone that's stuck with this problem. 
https://community.ultimaker.com/topic/42417-fix-for-connecting-creality-ender-3-neo-to-mac-via-usb/

Roos-AID, 2023-01-03:
Hi , thanks this works for me !
Suggest that Cura developer makes a change to their code to include this fix.

mstislavm, 2023-01-08:
> Hey All,
> 
> I found a workaround on the Ultimaker Community Forum for recognizing printers on Mac devices. Hope this helps someone that's stuck with this problem. https://community.ultimaker.com/topic/42417-fix-for-connecting-creality-ender-3-neo-to-mac-via-usb/

Thank a lot. The workaround works for me (MacOS 12.6 Monterey).

weaponrascal, 2023-04-24:
Hello All,  I would like to contribute my specific findings to this issue, and YES, my issue still causing me pain.
My Setup.
Apple Mac Studio with Apple M1 Max chip.  32GB ram, with Ventur1a 13.3 for Operating System
Ultimaker Cura 3D Slicing software, ver 5.3.0
Creality Ender 3 S1 3d printer running Ender3S1-F4-UBL Professional Firmware 2.1.3 20221002

I recently purchased the Apple Mac Studio (my MacBook Pro 2012 worked just fine).  Reason for the4 upgrade:  My autocad package (Fusion 360) was not longer function on the MacBook.
After getting the Mac Studio, installing OS Ventura, and handful of selected software, I stepped away for the past couple of months (only checking emails).
I recently wanted to reprint a project (discovered flaw), but when I began, I saw the same actions from my Ultimaker Cura as others.

The following has been suggested as work-around (on my MacBook Pro, it worked for me too).   But not on the Mac Studio.
Edit line 86 and change it from port_list = self.getSerialPortList(only_list_usb=True) to port_list = self.getSerialPortList(only_list_usb=False)   Editing True to False

When I look at my package contents for Ultimaker, I see this...
port_list = self.getSerialPortList(only_list_usb=Platform.isWindows())
Based on ther work-around, I changed "Platform.isWindows" to False
Side Note:  I also tried True in place of "Platform.isWindows"

End result was the same.

Command Line:  ls /dev/*usb*
My Result:  cu.usbserial-8340	tty.usbserial-8340   Note:  others stated getting 110, why is mine 8340?

One final note.  I discovered that if I shutdown Ultimaker Cura, then restart, getting to the Monitor tab, I can click the X Y Z jog, and it works.  Left, Right, Up, Down.  I can do this in varying patterns with no trouble.
Go to File, Open Recent, select file.  It loads into the Prepare tab.  Without doing anything else, I go to the Monitor tab, and NOTHING WORKS.  No PreHeat, No Jogging, Manual G-Code Send box has no reaction to the printer.

At every stage of the testing of suggestions by others, I reboot the MacStudio, restart Ultimaker Cura, Monitor tab functions as expected (jogging works), load .stl file, everything come to STAND-STILL.

I hope this info helps in resolving the issue.

bandito1, 2023-04-24:
@weaponrascal Did you installed CH34xVC driver for USB? I had a same problem with same setup, but when I installed that driver everything is fine. (Ventur 13.2.1, CURA 5.3.0)

[https://github.com/WCHSoftGroup/ch34xser_macos](url)

weaponrascal, 2023-04-24:
thanks for the assist bandito.  I made the attempt, but the link you provided did not work

No results matched your search.

so I went another route.  I did a search, found this:  
WCHSoftGroup / [ch34xser_macos](https://github.com/WCHSoftGroup/ch34xser_macos)

where I was able to download the :  ch34xser_macos   zip file.  Included wwere instructions, .dng, and a .pkg set of files.
I tried the .dmg file first, but it didn't perform the way I was expecting, so I did the .pkg, and it at the end says installed successfully.
First concern was this:

Chip Model Set
    CH340/CH341/CH343/CH9101/CH9102/CH9143 (USB to Single Serial Port)
    CH342/CH344/CH344/CH347/CH9103/CH9104 (USB to Multi Serial Ports)

For sure this was going to be an issue, I am running an Apple Mac Studio, Ventura 13.3.1 OS, and the only chipset I found in the System Report was :   BCM_4387

At any rate, I proceeded to follow the remaining instructions for install, found the tty.usbserial entry in terminal, made sure my package contents entries for Ultimaker and Creality slicers were back to their default.
Rebooted, started Creality slicer first, turned on the printer.  Monitor tab, displays NOTHING now.
Tried Ultimaker, same result.

Everything else appears to work. or at least it appears that way.\

I am now going to do a search to see if the are USB driver files for the Mac Studio M1 - then reinstall those (after I go through the process of uninstalling the other drivers.

bandito1, 2023-04-25:
@weaponrascal: Yo need to use .dmg file, because you running Ventura 13.x. That driver .dmg install is tricky, because not ordinary. First drop to Applications, after start it, install it, but need to accept in system preferences. If everything went good, you see two new usb (printer must be connected) when run this command in terminal window: tty.wch* 

Edit:
bad news is:
Everything is good, Cura see my Anet a8 printer, and I printed some parts trough USB. After I closed Cura. Today when I opened it again, not connected. All USB ports are ok, driver also. I reinstalled Cura 5.3.0, 5.3.1, but nothing happens.
It didn't work for long.

weaponrascal, 2023-04-25:
OK Bandito,  I performed another install o0f the .dmg file.  This time, it completed the way I expected, down to the accept in system preferences.
I then ran the terminal cmmand :  tty.wch*  ( as you indicated above).   Result:  Nothing found
So I ran a tried and true command :  ls /dev   Now I am able to see both the tty.wchusbserial8340  (do you know what the 8340 number is for???)
I am also seeing a :  cu.wchusbserial8340

So I tried to run both Cura and Creality.  Both programs start, but when I take a peek at the Monitor tab, it still won't recog...Won't recognize 3d printer.

bandito1, 2023-04-26:
> OK Bandito, I performed another install o0f the .dmg file. This time, it completed the way I expected, down to the accept in system preferences. I then ran the terminal cmmand : tty.wch* ( as you [...]

Now, I have a solution, and I know what is the problem. We have 4 pcs "USB ports", and all time when you connect the printer, USB ports randomly selected. I think Cura not scan all ports, so I use pronterface app, which is connect these ports and printer is connecting perfectly.

Steps:
1.) Make the gcode in Cura, save it.
2.) Open pronterface, connect your Anet a8
3.) Open your gcode model, and print it!

Pronterface: [https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0](url)

The biggest joke is, sometimes/randomly Cura connecting to printer also.

weaponrascal, 2023-04-28:
Bandito (and everyone else having this issue)

Here is what I found.   After doing a complete reinstall (fresh) of MACOS Ventura, without installing anything special, extra, printers, etc., I began my testing.

After setting wifi to my local software library, I installed Creality Slicer, edited the Package Contents, but still had the issue (or so I thought).  I'll explain this later.

So i installed Ultimaker Cura from my library, but since I the latest 3 versions, I installed ver 5.1.1.  Again, Package Contents changed, turned the 3d printer on, and all the items normally seen on the Monitor tab were there.
So i tried Z up, it worked.  Tried several axes movements and all worked.  Temp settings for the bed and nozzle also worked.

So I wanted to test pronterface, installed it as normal, but Package Contents for the USB printing file are not the same.  Will look into that as time permits.
But as I was moving from Cura to pronterface, I caught a glimpse of Creality Slicer, and the Monitor tab had too populated all the features applicable to that screen.  Delayed reaction.

So, ver 5,1 and Creality Slicer are functioning.  Will try Cura 5.2 tomorrow to see if the problem again surfaces.

As it stands now, the edited line 86 in file named : USBPrinterOutputDeviceManager
for version 5.3 needs a bit of work  and testing for the Apple MAC Studio M1 chip.
That line is:  port_list = self.getSerialPortList(only_list_usb=(isWindows******())
Since I am now running the 5.1.1 version, the specific text for line 86 is not visible, maybe the next version will have it (which means my Cura will stop working until i downgrade Cuira to 5.1.1)

bandito1, 2023-04-29:
> Bandito (and everyone else having this issue)
>
> Here is what I found. After doing a complete reinstall (fresh) of MACOS Ventura, without installing anything special, extra, printers, etc., I began my [...]

I found Cura 5.4 alpha, maybe help.
Here is the link: [https://drive.google.com/file/d/1dp2u-Pm0yb8ORewTjKSkg5A_qOGZuI4Z/view](url)

jimklo, 2023-09-18:
I'm having a similar issue. While I'm not on an M1, still on Intel.  I am on Ventura. Using Cura 5.4.0, with Marketplace distro of the USBPrintingNG v1.0.4 plugin.

I'm starting Cura from Terminal using the following incantation:

`CURA_BAUDRATE=115200 CURA_DEVICENAMES=".*usbserial.*" open /Applications/UltiMaker\ Cura.app`

Then looking at the `cura.log`, I see the following:

```
2023-09-18 15:47:16,966 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-AI02UM2H if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2023-09-18 15:47:20,739 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-1452110 if baud rate 76800 works. Retry nr: 1. Wait timeout: 15
2023-09-18 15:47:21,476 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-AI02UM2H if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2023-09-18 15:47:25,988 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-AI02UM2H if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
2023-09-18 15:47:30,517 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-AI02UM2H if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
2023-09-18 15:47:33,620 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]: Exception: Job <USBPrinting.AutoDetectBaudJob.AutoDetectBaudJob object at 0x1271a0820> caused an exception on worker JobQueueWorker [0]
2023-09-18 15:47:33,621 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]: Traceback (most recent call last):
2023-09-18 15:47:33,621 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:   File "serial/serialposix.py", line 500, in read
2023-09-18 15:47:33,622 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:     raise SerialException(
2023-09-18 15:47:33,622 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]: serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2023-09-18 15:47:33,623 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:
2023-09-18 15:47:33,623 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]: During handling of the above exception, another exception occurred:
2023-09-18 15:47:33,624 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:
2023-09-18 15:47:33,624 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]: Traceback (most recent call last):
2023-09-18 15:47:33,624 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:   File "UM/JobQueue.py", line 133, in run
2023-09-18 15:47:33,625 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:     job.run()
2023-09-18 15:47:33,625 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:   File "/Applications/UltiMaker Cura.app/Contents/MacOS/share/cura/plugins/USBPrinting/AutoDetectBaudJob.py", line 76, in run
2023-09-18 15:47:33,626 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:     line = serial.read_until(size = 100)
2023-09-18 15:47:33,626 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:   File "serial/serialutil.py", line 659, in read_until
2023-09-18 15:47:33,626 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:     c = self.read(1)
2023-09-18 15:47:33,627 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:   File "serial/serialposix.py", line 509, in read
2023-09-18 15:47:33,627 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]:     raise SerialException('read failed: {}'.format(e))
2023-09-18 15:47:33,627 - ERROR - [JobQueueWorker [0]] UM.JobQueue.run [135]: serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

During the process I see my printer (Wanhao i3) reboot during which I think all the exceptions occur.  I'm basically left in a state with no printer control.