Ultimaker/Cura #13734: Ironing and cooling

yet-another-average-joe, 2022-11-04:
### Is your feature request related to a problem?

More or less. I print PETG at "high" speeds, often long bridges, and have to activate the part colling fan @100%.
When ironing is activated, always with zero % flow (otherwise it produces a mess !), the cooling produces a matte surface. Deactivating part cooling produces a much shinier surface.

### Describe the solution you'd like

A fan % option for the ironing moves.
Ironing has some options, but not cooling. Cooling/not cooling while ironing is night and day when printing PETG.

### Describe alternatives you've considered

Editing gcode.

### Affected users and/or printers

All printers.

smartavionics, 2022-11-04:
Hi @yet-another-average-joe , that sounds like an interesting feature, I'll take a look at adding that capability to my next release. thanks.

MariMakes, 2022-11-08:
Hey @yet-another-average-joe,

Thanks for your report 🎉 
I agree with @smartavionics, this sounds like an interesting feature 😊 

I'll bring it up with our Print Profile Experts to see if this is something that can be implemented.

smartavionics, 2022-11-13:
FYI, I have added this feature to my latest release which can be found at https://github.com/smartavionics/Cura/releases

yet-another-average-joe, 2022-11-13:
Great !

GregValiant, 2023-01-28:
Look at #14394.  There is no special word in the UM Cura gcode for a surface that's being "ironed" so the "Skin" fan % would need to be set.