Ultimaker/Cura #13739: bug support arborescent

SONIC-GWADA, 2022-11-05:
### Application Version

5.0.0

### Platform

windows 10

### Printer

SOVOL SV03

### Reproduction steps

Bonjour je découpe le fichier 

### Actual results

j'ai régulièrement du manquant sur les supports arborescent, je dois jouer pendant longtemps sur l'angle pour trouver un qui valide le support partout, sinon je mets des supports manuel mais il n'imprime pas en mode arbre mais en mode normal et comme je mets les support  "en contact avec le plateau" cela ne fonctionne pas toujours

### Expected results

Avoir des supports sous les parties dans le vide

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

lien vers le .zip valable 14 jours
https://www.grosfichiers.com/udaDf64cNNr

MariMakes, 2022-11-07:
Hey @SONIC-GWADA,

Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.
Thanks for your elaborate report 👍

I've noticed that the behavior significantly improves if you reduce the Support X/Y Distance to 0.4
That makes me think it's a duplicate of https://github.com/Ultimaker/Cura/issues/8614
We have a ticket on our backlog with the intent to improve this behavior.

I love printing miniatures myself. I believe that if you rotate your model slightly it comes out way better and the support is easier to remove. 🤩 
[image]

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/8614

closed 2022-11-07 by MariMakes

SONIC-GWADA, 2022-11-07:
Hello
There were several projects. And depending on the type of figurine if rather simple I prefer to have as straight as possible to avoid the staircase effect.
I will try to reduce the XY but on the other hand it would be good to make sure that the manual supports are in tree mode and not in normal because on small parts it does not hold