Ultimaker/Cura #13740: It will not save the Z position in an projekt file! And do not delete old cure version or reuse the Material profiles from older Versions automatically

Solar320, 2022-11-06:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 3 V2

### Reproduction steps

I use the position tool to move the objekts, then **deselect** "drop down Model" to prevent automatic drop down.
Then I input 0.4mm for the Z parameter. If I save the project to disk and reopen the file all this parameters are set back to z = 0 and the "Drop down Model" is re enabled.

I want to save this parameter as I set it, only for such things the Projekt files exists, so it should save all paramters I set automatically.

The reason I want this is because I want to print an 0,4mm Layer of PLA and in the second Step I want to print ASA in an 0,4mm high over the PLA layer. So every Time I open the second project file I first have to move every Object to the right 0,4mm Z high and then save it to the Flashdisk. 

If I install a new Version of cura it do not unsinstall the old versions first, if I want this. Also it do not use my old Profile files, so I have to Export and Import it every Time I install a new version of Cura slicer.....

### Actual results

Cura slicer will reset my Parameters for no reason

### Expected results

It should save all parameters I set for the Project files. So I do not have to move all the Objects in the correct Z high after I reopen the Project file.
Also it seems to be not possible to select all Objects in one step so I can't move all objects at once to the 0.4mm Z high. 
Every time I reopen an project file I have to move every Object one after the other. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

A new Bug in there, I can't add an 3mf file to show you the problem, OMG.

GregValiant, 2022-11-06:
Thanks for the report.
You must "zip" project files to post them on Github.  We will need a project file because I can't duplicate the Z problem with 5.2.1 on Windows 10 and Ender 3 Pro.
I open a lot of project files.

sgielen, 2023-01-30:
Dear @GregValiant I seem to have the same issue as in the original report, where Z position of models is lost. Could you please reopen this?

I am running Ultimaker Cura 5.2.1 on macOS, printing on the Ultimaker 2.

This zip file contains three project files:
[hsnlamp.zip](https://github.com/Ultimaker/Cura/files/10535262/hsnlamp.zip)

- hsnlamp-4-24.stl is my 'primary' model, and I am trying to print it multiple times. Because it is standing up, and tends to fall over quite easily, I try to print it with some support models in between.
- driehoek.stl is that support model, it's a triangle with thin attachment points to the primary model so that it breaks off easily. 
- UM2_hsnlamp-4-24.3mf is the Cura project. I created and saved it last Wednesday. When I saved it, the "driekhoek.stl" models all had a nonzero height and "drop down model" disabled. When I just reloaded it, all the models had their "drop down model" setting re-enabled.

This is the printed version, from last Wednesday, working as intended:

[image]

Steps to reproduce:
- Create a new Cura project for the Ultimaker 2. I used 5.2.1 on macOS.
- Load hsnlamp-4-24.stl, duplicate it, and place them upright close to each other.
- Load driehoek.stl, disable 'drop down model' and place it such that its bottom point is "below" the bed (i.e. its footprint is larger), and with the two two points towards the middle of the two primary models.
- Duplicate driehoek.stl, disable 'drop down model' again (?) and place it so that its bottom point is inside the bottom driehoek, with the two points connecting to the middle of the primary model.
- Repeat the previous step a few times so that the print looks like the image above.
- Save the project and quit Cura.
- Reload the project and observe that all the driehoek.stls have dropped back down to the floor.

sgielen, 2023-01-30:
The same "drop down model" issue happens when grouping and ungrouping models. So, "drop down model" seems to be automatically activated (breaking the Z index) upon:
- loading project
- duplicating model
- grouping/ungrouping model

Perhaps the project issue is also resolved when this is fixed?

GregValiant, 2023-01-30:
No problem re-opening this.
When I load your project it looks as you say - skewed.

I'm on a Windows system so this may be Mac specific.  When I recreate the project using your instructions, save that project, and re-open it, it looks as it should.  So on my system I can't duplicate the problem.  I also can't really fix your original project file.  I'm attaching my project file so you can see if this works going from Windows to Mac as well.
[GV_Vase_Dominik_Cisar.zip](https://github.com/Ultimaker/Cura/files/10535595/GV_Vase_Dominik_Cisar.zip)

Someone from the Cura team will take a look.

sgielen, 2023-01-30:
@GregValiant Not sure if that's the project file you intended as it's a different model, but it looks like this when I open it:

[image]

If I lift it up (disable Drop Down Model and drag it up using the blue arrow), save it, then reopen it, it's dropped back down:

https://sjorsgielen.nl/cura.mov

GregValiant, 2023-01-30:
Geez.  Sorry about that.
That's a vase problem from the UM forum that I was working on.
I'll try again.
[GV_hsnlamp-4-24.zip](https://github.com/Ultimaker/Cura/files/10535741/GV_hsnlamp-4-24.zip)

sgielen, 2023-01-30:
@GregValiant I think this will remain open if you remove the “Needs Info” badge. I’m not sure if I’m allowed to do that.

GregValiant, 2023-01-30:
I'm not sure what's goin on there.  I had re-opened it and it should have stayed open for 2 weeks.  I did remove the label now.  We'll see.

sgielen, 2023-01-30:
@GregValiant when I open your .3mf file, it looks like two of the three `driehoek.stl` models are overlapping, namely the two I have selected here:

[image]

Indeed, all three of them have "Drop Down Model" enabled. What does it look like for you when you open the project?

GregValiant, 2023-01-30:
I inserted three.  In your image it looks like the top one may have slipped down into the middle one.  I did not enable "drop down model".  That looks like part of it, but I don't think it's all of it.
[image]