Ultimaker/Cura #13741: Building a Cura installer from source

yess1987, 2022-11-06:
### Application Version

5.2.0

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
git clone https://github.com/Ultimaker/Cura.git 
cd Cura
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
.\venv\Scripts\activate.ps1
python cura_app.py
conan create . cura/5.3.0-alpha@yess/testing --build=missing --update
conan install cura/5.3.0-alpha@yess/testing --build=missing --update -g VirtualPythonRunEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_install_info.json

1. Why the following format cannot be used:“cura/ 1.0.0@yess /testing”
2. Repeat the "conan create" and "conan install" commands several times, and the following commands appear.
   ERROR: Invalid generator 'VirtualPythonRunEnv'.

### Actual results

ERROR: Invalid generator 'VirtualPythonRunEnv'.

### Expected results

1. I can specify the version information by myself, instead of having to start with cura or a version larger than 5.2.0
2.I will modify the cura code many times. I hope I can execute "con an create" and "con an install" repeatedly to compile the cura executable program

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Repeat the "create" and "install" commands several times, and the following commands appear.

ERROR: Invalid generator 'VirtualPythonRunEnv'. Available types: txt, gcc, compiler_args, cmake, cmake_multi, cmake_paths, cmake_find_package, cmake_find_package_multi, qmake, qbs, scons, visual_studio, visual_studio_multi, visual_studio_legacy, xcode, ycm, virtualenv, virtualenv_python, virtualbuildenv, virtualrunenv, boost-build, pkg_config, json, b2, premake, make, deploy, markdown, GitHubActionsBuildEnv, GitHubActionsRunEnv, PyCharmRunEnv, VirtualPythonEnv, CMakeToolchain, CMakeDeps, MSBuildToolchain, MesonToolchain, MSBuildDeps, QbsToolchain, msbuild, VirtualRunEnv, VirtualBuildEnv, AutotoolsDeps, AutotoolsToolchain, BazelDeps, BazelToolchain, PkgConfigDeps, VCVars, IntelCC, XcodeDeps, PremakeDeps, XcodeToolchain, MesonDeps

"conan install . cura/@/ --build=missing --update"

" cura/@/" ，must start with cura？

MariMakes, 2022-11-07:
Hey @yess1987,

Thanks for your report 👍
So cool that you are joining the Building From Source Club 👑 
I'm still a fledgling member myself, and can't really help you. 😞 

@jellespijker could you take a look at this bug? 👼 

Thank you

julianCast, 2022-11-08:
Hi @yess1987, you should use the parameter `VirtualPythonEnv` instead of `VirtualPythonRunEnv` in your conan install.

yess1987, 2022-11-09:
Why do you need to download the libcharon source code during “conan install”? My network environment is bad, often don't have access to https://github.com/ultimaker/libcharon. Are there any good solutions?

  Cloning https://github.com/ultimaker/libcharon (to revision master/s-line) to c:\users\shuan\appdata\local\temp\pip-install-vhedagft\charon_52ee8219e66a4e37ae63e54dcdbc7815
  Running command git clone --filter=blob:none --quiet https://github.com/ultimaker/libcharon 'C:\Users\shuan\AppData\Local\Temp\pip-install-vhedagft\charon_52ee8219e66a4e37ae63e54dcdbc7815'
  fatal: unable to access 'https://github.com/ultimaker/libcharon/': OpenSSL SSL_read: Connection was reset, errno 10054
  fatal: unable to access 'https://github.com/ultimaker/libcharon/': OpenSSL SSL_read: Connection was reset, errno 10054
  error: unable to read sha1 file of .dockerignore (4f3f2b594684b252a4573296554263cae2db9108)

yess1987, 2022-11-12:
hi @julianCast, Why do you need to download the libcharon source code during “conan install ”?

jellespijker, 2022-11-12:
libCharon is one of the Python dependencies of Cura (and Uranium) they are specified in requirements-ultimaker.txt pointing towards our Git repo. The VirtualPythonEnv generator installs these automatically, Since they're needed to run Cura.

If your connection really is that unstable you could try to remove that line from the above mentioned file, so it (libCharon) doesn't get installed automatically.

mind you would then have to install libCharon manually in the Python environment

yess1987, 2022-11-13:
git clone https://github.com/Ultimaker/Cura.git
cd Cura
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv -----Did   it (libCharon) do n't get installed automatically?
.\venv\Scripts\activate.ps1

I want to create my custom build with changes on my local computer. Do I have to install libCharon manually in the Python environment again?
conan create . cura/5.3.0-alpha@yess/testing --build=missing --update
conan install cura/5.3.0-alpha@yess/testing --build=missing --update -g VirtualPythonRunEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_install_info.json

jellespijker, 2022-11-13:
@yess1987 I think it will be quicker if we do a call and share screens. Because I think some things are lost in translation.

Would you appreciate a video/screen sharing call?

CHSIVASANTHOSH, 2022-12-20:
@jellespijker  if you are able to build the cura from source so that you are able to buld custom exe & cli.exe files then would be glad that if you can make a video on your computer as I got my head to flame to get this done in the ways possible

CHSIVASANTHOSH, 2022-12-20:
@all if anyone are able to build the cura from source so that you are able to buld custom exe & cli.exe files then would be glad that if you can make a video on your computer as I got my head to flame to get this done in the ways possible