Ultimaker/Cura #13744: Slicing bug

slpatterson62, 2022-11-06:
### Application Version

5.02

### Platform

windows 11

### Printer

Anycubic Mega X

### Reproduction steps

I am trying to slice items from a list of items for a particular project. I have printed it before this with no hassle. I keep getting a message of a "slicing bug" and am unable to slice items previously sliced the same day! I press slice, it starts and then stops with this error message

### Actual results

Nothing. i can try to slice again, even after exiting and opening a new window. I still get the "slicing bug message

### Expected results

it should slice them i download to a memory card which i insert into my printer and print

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[AI3MX_ARM_R.zip](https://github.com/Ultimaker/Cura/files/9946013/AI3MX_ARM_R.zip)

MariMakes, 2022-11-07:
Hey @slpatterson62,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

[image]

MariMakes, 2022-11-08:
I'm happy to hear that your issue is resolved.
I'll be closing this ticket. 
Good luck printing! 💪

closed 2022-11-08 by MariMakes