Ultimaker/Cura #13745: Solid bottom when there is a round hole

IvetalI, 2022-11-06:
### Application Version

5.2.1

### Platform

Windows

### Printer

Custom

### Reproduction steps

Slicing in vase mode a vase with a round hole at the bottom.  

### Actual results

Solid bottom layers without hole.

### Expected results

Bottom layers with a hole

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9946162/cura.log)
[Bell_twisted.zip](https://github.com/Ultimaker/Cura/files/9946163/Bell_twisted.zip)

GregValiant, 2022-11-06:
Thanks for the report.
There is something wrong with your project file and although Cura opens it, there is no model.  Could you try again?  Just the STL will work as I have a feeling that the model isn't suitable for Spiralize as holes interfere with the toolpath and are not allowed.  A proper "vase mode" model is solid through-and-through with only a single exterior surface.

smartavionics, 2022-11-07:
>  A proper "vase mode" model is solid through-and-through with only a single exterior surface.

This hasn't always been the case, it's a Cura 5 regression.

IvetalI, 2022-11-07:
The model is fully appropriate for vase mode. I used Cura 4.xx and it sliced normally.

GregValiant, 2022-11-07:
I still don't get a model in the 3mf file but your "LeVi volcano vase 02" profile came in.

4.x versions seem to ignore any inner walls where 5.x versions develop toolpaths for them.  It is certainly different from one to the other. "Remove all Holes" appears to be enabled by default in 5.2.1 (where it isn't in 4.13.1).  That would cause what you are seeing but the model will not Spiralize without it in 5.2.1.

This is 4.13.1.  It's my own model with a through hole.
[image]

This is in 5.2.1 with "Remove All Holes" turned off.
[image]

In 5.2.1 the "Inner Wall" of the model is being sliced and I know that the gcode will reflect a spiral path around the outside loop, extrude to the inside loop and move around it, and then "travel" back to the outside.  Although the Z does increment upward all along the path, it's hard to call this "Spiralize" with those bridge lines running across the print.
[image]

I have no opinion on this issue but I will note that 4.13.1 appears to be ignoring a feature of the model (the inside of the goblet) while 5.2.1 does not.  That's part of the reason I said "Not a proper model for spiralize".  I believe that software should not ad-lib.

@smartavionics has a **[Cura fork](https://github.com/smartavionics/Cura/releases/tag/4.20.7)** based on the UM Cura 4.x version.  It does an excellent job in vase mode and has a few other tweaks.  You don't have to un-install your UM versions to run it.

I'll leave this as a bug report but it may be changed to a feature request by the Cura Team (something like "Return to previous Spiralize behavior") at some point so I will add the "Improvement" label.
Although this is "expected behavior" in 5.2.1, a lot of people hate it.

GregValiant, 2022-11-09:
I'll go ahead and remove the bug label but I'll leave this open as an "Improvement".  There has been a lot of griping about the way 5.x handles Spiralize.