Ultimaker/Cura #13746: Can't use settings in marketplace because the menu gets deactivated before you can click on it. (bad menu placement)

SomeDeveloper13, 2022-11-06:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Click Market place.
2. Click settings gear icon.
3. try to move mouse to menu to click "manage packages", but menu disappears before you can click.

### Actual results

try to move mouse to menu to click "manage packages", but menu disappears before you can click.

### Expected results

I'm able to click on and enter the manage packages menu.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

MariMakes, 2022-11-07:
Hey @SomeDeveloper13,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Is this the issue you are encountering? 
[image]

I can bring it up to the team, but I fear that they will not have time to work on it any time soon. 

Fingers crossed 🤞

MariMakes, 2022-11-09:
Hey @SomeDeveloper13,

We've added a ticket to the backlog with the intent to take a quick look if we can improve this. 
For internal reference CURA-9874

Thanks for the report! 👍