Ultimaker/Cura #13747: Hide Seam shows seam.

tuddenha, 2022-11-06:
### Application Version

5.1.2

### Platform

Windows 8.1

### Printer

Iain’s BQ Prusa i3 Hephestos

### Reproduction steps

1. Set Z Seam Alignment to Sharpest Corner.
2. Set Seam Corner Preference to Hide Seam.
3. Sliced.

### Actual results

The Z Seam was put on an outside corner (neither the sharpest nor the bluntest), despite inside corners being available.

### Expected results

The Z Seam should have been put on the sharpest inside corner.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[smaller_filament_clip 5.1.2.3mf.zip](https://github.com/Ultimaker/Cura/files/9946620/smaller_filament_clip.5.1.2.3mf.zip)
[image]
[image]

alperozoner, 2022-11-06:
Yeah this has been an issue since 5.1.1. I hope it gets fixed on 5.1.3.

GregValiant, 2022-11-06:
Thanks for the report.
Below is the Cura Prepare View of your model (with the bad surfaces shown in red) and the same model after repair.  When a model has flipped normals like this - the slicer has trouble telling the inside from the outside and consequently mistakes can be made.  In addition, the chamfers on the filament holding diameter are causing very thin sections and I think Cura may be seeing them as separate features instead of as part of the whole piece.  
[image]

The first thing is to repair your model and get those flipped normals facing inward.
I think you need to go over your settings as well.  It's a very small clip and settings like "Initial Layer Horizontal Expansion" at -.322 make a big difference.  The first layer looks like about 4 separate pieces.  In Mesh Fixes you have "Remove Mesh Intresection" enabled as well as "Alternate Mesh Removal".  Those certainly aren't required for this model.
Your final printing temperature is 165°.  That's actually 5° below the standby temperature for PLA and is 45° below my printing temperature.  If it is working for you that's fine, but it is quite low.

There may well be a bug here s there have been other reports of odd behavior with the Z seam but between the bad model and some odd settings it's hard to tell one way or the other in this instance.  If you would repair the model and take a look at your settings you may get different results.

MariMakes, 2022-11-07:
Hey @tuddenha,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

@alperozoner is correct. I just checked a build with the fix of https://github.com/Ultimaker/CuraEngine/pull/1750 and your zseam is appearing int he inside corner as you descibed. So keep an eye out for the next release 😉 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

tuddenha, 2022-11-07:
> I just checked a build with the fix of [Ultimaker/CuraEngine#1750](https://github.com/Ultimaker/CuraEngine/pull/1750) and your zseam is appearing int he inside corner as you descibed. So keep an eye out for the next release wink
> 
> To me, it doesn't seem like a bug in Ultimaker Cura anymore. Would you still like to keep this issue open or do you agree that we can close it?

Thank you, MariMakes.  If it’s fixed in the latest engine, I agree you can close it.

closed 2022-11-07 by MariMakes