Ultimaker/Cura #13749: Cura 5.2.1 fail to slice model

philbowers, 2022-11-07:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality CR-10 V2

### Reproduction steps

1. Load .stl
2. Press slice
3. Failed report a bug

### Actual results

Failed to slice

### Expected results

Model to slice and generate gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Fresh install of Cura 5.2.1

[cura.log](https://github.com/Ultimaker/Cura/files/9947476/cura.log)
[Buddy.zip](https://github.com/Ultimaker/Cura/files/9947483/Buddy.zip)

MariMakes, 2022-11-07:
Hey @philbowers,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I can slice your STL just fine on my Ultimaker S5, so it's not an issue with the model.
[image]

But in your logs I can read CuraEngineBackend.CuraEngineBackend._terminate [369]: Engine process is killed. Received return code 3221225477

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

philbowers, 2022-11-08:
It sliced when I switched to my ender 3 it sliced just fine. Besides the obvious differences, the main difference between the two are.
Ender 3 - .4mm nozzle, upgraded 32bit mks board with 2209 drivers and, controlled with Octoprint 
CR10 - .6mm nozzle, stock creality board and, is controlled with Klipper (Fluidd)

[CCR10_Buddy_3mf.zip](https://github.com/Ultimaker/Cura/files/9963493/CCR10_Buddy_3mf.zip)

MariMakes, 2022-11-09:
Hey @philbowers,

Thanks for sharing your files 👍

When I change your Adaptive Layers Maximum Variation from a negative to a positive number I'm able to slice your printjob.
[image]

Why do you wnat this value to be negative? 

I checked the setting defition and I don't see any maximum or minimum values for this setting. 
Probably since it's experimental.

 I'll bring it up to our print profile experts to take a look to apply proper warning values here. 👍 
 
To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

philbowers, 2022-11-10:
I see what you are saying but, I do not know where that value is stored/ changed at. And oddly it only appears in that profile.
[image]
[image]
[image]
[image]
[image]

philbowers, 2022-11-11:
I do not know how that fetcher was enabled, as it wasn't selected as enable when I found it. I have set it to zero and left it disabled which, fixed the problem. I give you My Sincerest Thank You for helping solve this issue and, will be closing this issue out.

closed 2022-11-11 by philbowers