Ultimaker/Cura #13750: this is a thing that happened?

rprussell2004, 2022-11-07:
### Application Version

Arachne Engine Beta 2

### Platform

Windows 10

### Printer

Anet A8 Plus w/Klipper

### Reproduction steps

1. Load Model
2. Slice model

### Actual results

"Unable to slice. Submit a bug."

### Expected results

Seriously?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9947670/cura.log)
[buggy bug bug.zip](https://github.com/Ultimaker/Cura/files/9947674/buggy.bug.bug.zip)

MariMakes, 2022-11-07:
Hey @rprussell2004,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I'm able to slice your project file in Cura 5.2.1.
We made a number of improvements since Arachne Beta 2, and resolving a number of crashes was one of them.

[image]

You can download the latest version of Cura here:
https://github.com/Ultimaker/Cura/releases/tag/5.2.1

If that's not an option for you, can you share why you are hesitating to upgrade?
Thank you 🙏

rprussell2004, 2022-11-07:
SWEEEEEEEET thank you very much!

I was able to slice it in 5.1.1 also, I'll go get the new one.

I had thought the 'arachne' engine was the new bestest thing?

closed 2022-11-07 by MariMakes

RobertRussell-opusivs, 2022-11-07:
Y'all rock. Seriously. I love this software. Thanks again!