Ultimaker/Cura #13751: Is "Raft Air Gap" ineffctive?

844748699, 2022-11-07:
### Application Version

5.2.1

### Platform

macOS 13.0

### Printer

Ender-3 V2

### Reproduction steps

- added support and raft
- set "Raft Air Gap" to 0

### Actual results

there is a gap between the top of the raft and the bottom of my print

as you can see below, layer 4 contains raft
layer 5 only contains support and doesn't contain my print
layer 6 contains my print

[image]
[image]
[image]

### Expected results

there should be NO gap between the top of the raft and the bottom of my print when "Raft Air Gap" is set to 0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

MariMakes, 2022-11-07:
Hey @844748699,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

We've been investigating a number of raft issues like 
https://github.com/Ultimaker/Cura/issues/13713

Can you share your project file so we can see if it's related? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

844748699, 2022-11-07:
file is in this link:
https://pan.baidu.com/s/1Xl0FYBtcZORPeNV5bJYtLA?pwd=1234

MariMakes, 2022-11-08:
Hey @844748699,

Thanks for sharing the files.
I did not have access to this file sharing site, but with some help I was able to retrieve the file.

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13713
We have a ticket on our backlog with the intent to improve this behavior.
Thanks for this report, that is going to improve our ability for troubleshooting 💪 

https://user-images.githubusercontent.com/40423138/200557691-5c2d69c5-116a-4929-955e-7545c4ac3b9d.mp4

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13713
Thank you 🙏

closed 2022-11-08 by MariMakes