Ultimaker/Cura #13753: Slicer making seemingly useless move

rondlh, 2022-11-07:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Load the attached project (2 cylinders. Cylinder 1: R=175mm. Cylinder 2: R=174.4mm Cutting mesh) Creating a narrow band of 0.6mm (=nozzle diameter).
2. Watch the final nozzle move of each layer

### Actual results

I see this happening.
1. The nozzle moves clockwise in a circle starting at 9 o'clock.
2. When back at 9 o'clock, retract, move to about 5 o'clock, end retract, DO NOTHING, retract and go back to 9 o'clock for the next layer. In the gcode I can see that except for the retract and Z-hop nothing is happening at 5 o'clock. The move there seems useless.

**Here the gcode with my added comments:**
G1 X25.344 Y199.817 E1697.29581
G0 F1620 X25.338 Y200.016
;CIRCLE IS FINISHED
G1 F1500 E1694.29581
G1 F360 Z5.4
;Retract 3mm and Z-HOP 0.8mm
G0 F14400 X341.311 Y97.284 Z5.4
G1 F360 Z4.8
G1 F1500 E1697.29581
;Go to about 5 o'clock, end Z-hop, end retract
G1 F2136.5 X341.311 Y97.284
**;THE NOZZLE WAS ALREADY IN THIS LOCATION!!!**
G1 F1500 E1694.29581
G1 F360 Z5.4
;Retract 3mm and Z-hop
;MESH:NONMESH

### Expected results

I expected to not make seemingly useless moves.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Circle Test.zip](https://github.com/Ultimaker/Cura/files/9949421/Circle.Test.zip)
Attached the Cura project, a preview image of the scene, and the created gcode.

MariMakes, 2022-11-07:
Hey @rondlh,

Thanks for your report 👍
This one had me scratching my head for a while.

But the reason this move is made is because the zseam is defined in that position.
If you change the zseam to Front Left your move looks like this:
[image]

In your project this z seam position is defined as sharpest corner, but that's a bit hard to decide with a perfect circle. 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

rondlh, 2022-11-07:
> Hey @rondlh,
>
> Thanks for your report 👍 This one had me scratching my head for a while.
>
> But the reason this move is made is because the zseam is defined in that position. If you change the zseam to [...]

@MariMakes Thanks for looking into this. I can understand your explanation, still it's a completely useless move, nothing is happening at the 5 o'clock point, right?
Also, if I change the diameter of the outer cylinder a bit from 350.6 to 350.8mm, then this issue disappears.
Either way, thanks, I will close the issue soon...

MariMakes, 2022-11-07:
Hey @rondlh,

When working on a different issue I realized, we also have a Shortest Z Seam Alignment that is made exactly for this purpose!
If you select it it will minimize the length of the travel moves leading towards the seam. 😮 In that case you don't need to define the zseam location at all. 

[image]

rondlh, 2022-11-08:
Thanks, so many options, Cura is really an amazing piece of software! The improvements in version 5 are great, they really helped me a lot. Thanks for your great support!

closed 2022-11-08 by MariMakes