Ultimaker/Cura #13757: Build from Github Actions

julianCast, 2022-11-07:
### Application Version

5.2.1

### Platform

MacOS

### Printer

None

### Reproduction steps

Simplest way to build Cura 5

Reading the [documentation to build](https://github.com/Ultimaker/Cura/wiki/Build-With-GitHub-Actions) Cura 5 I came to the conclusion that I'd need a jfrog ultimaker account to run the Github actions, as it follows:
> CONAN_USER: GitHub service user account on ultimaker.jfrog.io e.q.: svc_conan@ultimaker.com
> CONAN_PASS: Github service user account token for ultimaker.jfrog.io e.q.: KJGHIHJKLUHGUIYGUIOKLJ

If that's so, how do external developers build their own Cura version? Am I missing something?

For example, if I only modify Uranium and Cura on my local environment and I want to release them, is there a way to make a build without using Github Actions or jfrog?

Thank you!

### Actual results

Does not apply

### Expected results

Being able to build without Jfrog account/github actions.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Does not apply

MariMakes, 2022-11-07:
Hey @julianCast,

So awesome that you are joining the Run From Source Club 👑 
You are not required to have a jfrog account as a community developer. 

If I understand it correctly you can run the command
`conan remote remove cura`
To remove the part of cura that refers to these credentials and the need for having credentials. 

If that doesn't make sense we can ask @jellespijker to explain it better. 🤞 

Good luck!

julianCast, 2022-11-08:
Hi @MariMakes ! Thanks for replying.

Yes, I have no problem Runing from Source. My question above is about the Build (AppImage, dmg and nsis installers).

MariMakes, 2022-11-11:
Hey @julianCast,

I apologize, our documentation for actually building for building (AppImage, dmg and nsis installers) is not up to date yet. 

You can look into our GitHub Action as inspiration
https://github.com/Ultimaker/Cura/blob/c9fc171776ae96cc1b8b17acc522ff5aa9a9f4fc/.github/workflows/cura-installer.yml
and https://github.com/Ultimaker/Cura/blob/main/packaging/dmg/dmg_sign_noterize.py

I'm super curious. What do you want to achieve by creating a Build? 
Are you intending to share it with others or are you using it for research purposes? 
I would love to know more about your project, maybe we can help.

julianCast, 2022-11-15:
Hi, I'm currently working on implementing changes for research purposes. So by creating a build I'd know that I'd be able to easily distribute it without having to deal with jfrog accounts. 

I was actually able to create it following the Github action you've mentioned, so thank you @MariMakes  for that.

What I'm still working on is, including local uranium content (instead of from a repo).

jellespijker, 2022-12-13:
great to know that you succeeded with creating a build. I will close this issue as completed. 

Please take note of license of Cura, Uranium and CuraEngine especially if you're (re-)distributioning  it.
It stands to reason we are also Curious about the changes you make,  certainly if  it is done in a research project. Mostly because we are all science needs here ;-)

closed 2022-12-13 by jellespijker

jwesel, 2023-12-27:
> I was actually able to create it following the Github action you've mentioned, so thank you @MariMakes for that.

How were you able to build it?