Ultimaker/Cura #13759: Unable to slice with support interface enabled.

SassyWassy, 2022-11-07:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Anet ET4 X

### Reproduction steps

1. Load the model into the program.
2. Enable support interface.
3. "Unable to slice with the current settings. the following settings have errors: Support interface Horizontal Expansion, Support roof horizontal expansion, Support floor horizontal expansion.

### Actual results

Error (Obviously)

### Expected results

Successful slicing.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Loaded on a simple model, but occurs no matter what model is used. It's likely some minute detail in the settings menu.

[image]

[image]

[AET4X__3DBenchy.zip](https://github.com/Ultimaker/Cura/files/9956148/AET4X__3DBenchy.zip)

MariMakes, 2022-11-08:
Hey @SassyWassy,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

I took a look at your file. Your values are red because they are influenced by the setting Support Horizontal Expansion. 
If you briefly change your support back to normal, and change the Support Horizontal Expansion to 0 instead of -2, you can change back to tree support and your printjob will slice:
[image]

https://user-images.githubusercontent.com/40423138/200563013-8c5f12cb-0461-483f-9ab3-6cb13079c74b.mp4

You can download the new version here: 
[Mari_AET4X__3DBenchy.zip](https://github.com/Ultimaker/Cura/files/9961040/Mari_AET4X__3DBenchy.zip)

To me, it doesn't seem like a bug in Ultimaker Cura, but an issue with setting choice. 
Would you still like to keep this issue open or do you agree that we can close it?

SassyWassy, 2022-11-08:
So sorry about that, I tagged the issue wrong. Thanks for the help!

closed 2022-11-09 by MariMakes

jmger, 2022-11-21:
I am running CURA 5.2.1. I am getting the same error message. I want to try the Support Interface expansion option. The problem I am having is this... I am using Tree Supports. The supports only print to the edge of my overhang. Therefore the edge of the overhang does not completely print fully. I was hoping that extending Tree Supports a little would solve this issue. Thank You.

MariMakes, 2022-11-28:
Hey @jmger,

Do you mind opening a new issue for us? 
This issue is already closed and I believe your issue is slightly different than the issue from SassyWassy. 

Thank you 🙏