Ultimaker/Cura #13764: Feature idea - sort plugins/material profiles with available updates as priority at top of list

tastyratz, 2022-11-08:
### Is your feature request related to a problem?

When you have many plugins or materials installed and have updates available, you need to scroll the list and try to find what you have that has available updates

### Describe the solution you'd like

I think available updates in the marketplace should automatically appear at the top of the list to make updating easier.

### Describe alternatives you've considered

Scrolling and finding the update the long way

### Affected users and/or printers

all users

### Additional information & file uploads

This is a minor and low-priority suggestion with some quality of life gain.

MariMakes, 2022-11-09:
Hey @tastyratz,

Thanks for your report 👍

If I understand correctly you want in this case to have the settings guide ordered on top. 
[image]

I recall that it was planned to implement this feature for the upgrade in Cura 5.0 but it was cut since we ran out of time.
I can check again with the team, but I predict that it will be deferred to focus on other things.

tastyratz, 2022-11-10:
Also to tack onto that, it would be useful to group and maybe filter the main plugin list by installed/ not installed on top of the ones requiring updates. As it sits today it's a big free for all list intermixed. It's not sorted alphabetically.

I understand I can click settings and manage plugins to see installed ones, the general marketplace UI is a bit unclear. That gear seems to only be an installed page, so, maybe it should just be another square for installed plugins?

I also just noticed that the little square with an arror intended to open the plugin page is only functional on the plugin primary page but dead on the manage packages page. It's also odd they are called packages there and plugins on the main page.

Overall, still no kind of priority here, especially in contest to time on functional fixes and improvements - just general observations on UI.