Ultimaker/Cura #13765: (5.2.1) Printing temperature initial layer changed before finishing initial layer

aguuag, 2022-11-08:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Artillery

### Reproduction steps

1. Set a Printing temperature for initial layer different from Printing temperature
2. Slice

### Actual results

First layer starts printing at "printing temperature initial layer" but then temp. changes to "printing temperature" before finishig first layer (at line n 200 in example gcode attached)

### Expected results

Pringing temperature initial layer is maintained until first layer finishes

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

gcode and project files attached
[TestTempInitLayer.zip](https://github.com/Ultimaker/Cura/files/9965109/TestTempInitLayer.zip)

GregValiant, 2022-11-09:
Thanks for the report.
I agree that's not where the M104 line should be.

If you load the Printer Settings plugin from the Cura Marketplace you will get access to some of Cura's hidden settings.  Two of them are "Heat Up Speed" and "Cool Down Speed" for the nozzle.  Cura uses those settings to calculate where the M104 line needs to be in the gcode so that the temperature reaches the set point right at the layer change.  For both of those settings the default is 2°/second.  With a temperature difference of 5° the M104 line should be very close to the LAYER:1 line in the gcode as it would only need 2.5 seconds to drop from 205 to 200..

What I found was that with your "Test w/Supp" profile active the M104 line is placed about 750 lines before the LAYER:1 line.  I could not get Cura to move it with that profile active.

With the profile changed to your "Standard Quality" the M104 line is placed 11 lines before the "LAYER:1" line in the gcode.  That's where it should be.  As another test I switched to my Ender as the active printer and with one of my custom profiles active,  There was no problem with it either.  Adjusting the "Cool Down Speed" moved the M104 line within the gcode file with your printer active and the Standard Quality profile active and also with my printer active.  In both of those instances it worked as it should.

So it looks like there is a problem within your "Test w/Supp" profile.  I have no idea what it could be.  I did check the definition files for both your printer and extruder and they looked fine.

I'll leave the bug label on this so someone from the Cura team will take a look but it looks like something within that profile is causing the problem rather than a Cura bug.

As a workaround, load the "Standard Quality" profile and change your line widths and and other settings so it matches the setup of that "Test w/Supp" profile.  Then click on the profile name and down at the bottom is "Create profile from current settings/overrides".  Use that to make a new custom profile and discard that "test w/supp" profile.

aguuag, 2022-11-10:
Hi thanks for your reply. 
After reading your comments, I tried the parameters one by one, and I discovered that the one that is causing the issue is the "top/bottom pattern". When I change it from "lines" to "zig zag" then the M104 moves up in the gcode file. This happens both in the "standard quality" profile and in the "test w/supp" profile
I hope this could help finding the bug.

GregValiant, 2022-11-10:
I think you found it.  It sure seems to be the ZigZag pattern causing the problem.
The "Final Printing Temperature" still doesn't go in though.