Ultimaker/Cura #13768: Fuzzy Z

mattwn, 2022-11-08:
### Is your feature request related to a problem?

"Fuzzy Skin" helps to both hide layer lines and provide surface texture. It is currently limited to the XY plane. The problem is that you can't get the benefit of hiding layer lines without increasing surface roughness.

### Describe the solution you'd like

A "Fuzzy Z" option which only dithers Z rather than XY could help to visually hide layer lines without an excessive increase in surface roughness. 

User selectable parameters would be: 

- minimum and maximum height of the layer (on the order of 0.05 mm min and 0.35 mm max with a 0.4 mm nozzle),
- rate of dither (with faster rates resulting in a tighter pattern and slower rates resulting in a pattern that might loosely resemble a voronoi organic cellular wall).
-maximum allowable height deviation (limited by the users nozzle geometry)

### Describe alternatives you've considered

To reduce layer line visibility, we typically increase fuzzy skin and decrease layer lines, but both have negative effects on surface roughness and print time respectively.

### Affected users and/or printers

Anyone seeking an improvement is surface quality through the hiding of layer lines.

Anyone intentionally seeking a cellular look to their organic models

### Additional information & file uploads

Conical slicing, and other "true 3d" slicing is an active area of research. This seems like a potential low effort way to bring true 3d path planning to Cura.

The lowest level of effort to implement this might be to limit it to spiralize mode only as a start. The layers would have to be sliced serially rather than in parallel so that the prior z height was known.

[image]

GregValiant, 2022-11-09:
That would be a seriously fancy bit of programming.  Keeping the flow consistent would probably be impossible.
Fuzzy only applies to the outside wall so inner walls (right behind the fuzzy wall) are straight.
3mm humps on the top skin surface would be unsupported and the nozzle would crash into them on the next pass?  I think you would just have a bunch of loose sausages barely adhered to the skin.
Good luck.  It's an interesting proposal.

mattwn, 2022-11-10:
I agree your concerns are completely valid for solid models.

I think a much simpler starting point would be to limit it to spiralize slices only. Then there are no inner walls to worry about. Spiralize also already has the limitation of no top skin.

mattwn, 2022-11-10:
This might just be me playing favorites, but I also thought it would be cool if Cura could claim to be the first mainstream slicer that offered some form of true 3D slicing.

GregValiant, 2022-11-10:
The patents for non-planar slicing make trying to get into it a financial challenge.  Either no one has really figured out how to do it, or the hardware is very expensive, or there really isn't a market for it yet
I think AutoDesk holds a main patent on the process.  I don't think they've released any software for the process.  It's kind of like the Creality belt printer.  Interesting idea but how many swords do you need to print.