Ultimaker/Cura #13769: Ender-3 S1 Default Monotonic Ordering.

BakeJake, 2022-11-08:
### Is your feature request related to a problem?

I just found out after a 15 hour print that monotonic ordering was off by default for the Ender-3 S1 Profile.
Luckily I enabled monotonic ironing hoping it levels out the discrepancy!!

### Describe the solution you'd like

Monotnic Enabled by default.

### Describe alternatives you've considered

None, It just needs to be enabled :)

### Affected users and/or printers

Creality Ender-3 S1

### Additional information & file uploads

Thanks In advance.

GregValiant, 2022-11-09:
That would need to be done in the definition file of your printer.  Since it's Creality printer a contributor would need to make the change in the overrides section and submit a pull request.
You can save custom profiles with Monotonic enabled.

BakeJake, 2022-11-09:
Thank you! I'll have to go check that file, I have made a custom profile actually as of now I just like using the default and making my changes lol. when doing different prints and disregarding all overrides I lose things like monotonic and guess forgot to set it again on that print!

MariMakes, 2022-11-09:
Since Cura is open source, you or anyone else is more than welcome to work on enabling this setting by default and create a pull request yourself. 

You can find the files you need here: 
https://github.com/Ultimaker/Cura/tree/main/resources/definitions
And here
https://github.com/Ultimaker/Cura/blob/main/resources/definitions/fdmprinter.def.json

If you are stuck in any way, please reach out! I'll be happy to help out.