Ultimaker/Cura #13770: cura software

letsgo101, 2022-11-09:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

I just downloaded this cura software and then I got it figured out how to change the files and now when I try it tells me that modifications are no longer available also it no longer works to download stl files it switches them to something else. When I was going through my laptop, I was deleting files and I'm not sure i could have deleted something I wasn't supposed to I'm not sure. 

### Actual results

I can't get anything to change to the right file

### Expected results

Before i could run it through my software and make it from an stl file to a gcode file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing else i can think of

GregValiant, 2022-11-09:
_"When I was going through my laptop, I was deleting files and I'm not sure i could have deleted something I wasn't supposed to I'm not sure."_
This doesn't really sound like a Cura bug so I will remove the label.
Did you try re-installing Cura?

MariMakes, 2022-11-10:
Hey @letsgo101,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Did you try reinstalling Cura? 

If that doesn't work... 
Do you maybe have a screenshot of what you are seeing? 
Or can you phrase it as; I'm seeing ..... but I expected to see .....
Or can you share how you are loading your STLs in Cura?

Thank you 🙏

closed 2022-11-25 by github-actions[bot]