Ultimaker/Cura #13771: wont splice

Kingkongg22, 2022-11-09:
### Application Version

5.0

### Platform

mac

### Printer

ultimaker

### Reproduction steps

splice then print

### Actual results

keep saying to report bug

### Expected results

should splice

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

wont splice

GregValiant, 2022-11-09:
Thanks for the report.
@MariMakes handles all splicing reports here.  She will be happy to help you with this.
What model printer do you have?

MariMakes, 2022-11-09:
Hey @Kingkongg22,

Welcome to the Ultimaker Cura Github 🚀 

@GregValiant is right, we are here to help 😊 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2022-11-23 by github-actions[bot]