Ultimaker/Cura #13773: ERREUR INCONNUE SLICE

JOLLIVON, 2022-11-09:
### Application Version

5.2.1

### Platform

WINDOWS 11

### Printer

ENDER 3 V2

### Reproduction steps

SLICE

### Actual results

ECHEC DE LA DECOUPE AVEC UNE ERREUR INCONNUE

### Expected results

DECOUPE
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

FILE
[CE3E3V2_Corps1.zip](https://github.com/Ultimaker/Cura/files/9968852/CE3E3V2_Corps1.zip)

JOLLIVON, 2022-11-09:
> Hey @JOLLIVON <https://github.com/JOLLIVON>,
>
> Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.
>
> Welcome to the Ultimaker Cura Github 🚀
> Thanks for your report 👍
>
> I tried to slice the file you shared on Cura 5.2.1 and I was able to slice
> it succesfully. 😮
> [image: image]
> <https://user-images.githubusercontent.com/40423138/200796893-3c975659-7ab3-4399-8045-1e8bb2794d8c.png>
>
> Can you check for me if you shared the correct file? 🤔
> Thank you 🙏

Thanks .
i think it's not a problem of the part because i had this kind of error
many times with the 5.1 and i succes to slice sometimes. maybe it's a
problem of the installation of the software.
I tried differents part and i have the same issue.
When i have this issue i can't slice anything. so i use ideamaker to slice
but i prefer Cura.
I don't understand this issue if i can help to solve it.
Best regards.
Jérôme Ollivon

*Mail*: ***@***.***
*Téléphone:* +33 6 81 55 04 10
*Adresse: *4 allée des boulards
                   78430  Louveciennes

MariMakes, 2022-11-09:
Thanks for the update!

Ah, than it might be related to something on your device. 
Than it would be more helpful for us to see your logs the next time you encounter this. 

You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

JOLLIVON, 2022-11-09:
> I don't have this folder in my roaming.
> for information i have also creality slicer 4.8 for my CR30 belt printer.
> Maybe there is an interaction between the configuration file of the two
> slicer ?
> I restarted my computer and the slice worked.
> I will try the process to generate the issue
> *Jérôme Ollivon*
>
> *Mail*: ***@***.***
> *Téléphone:* +33 6 81 55 04 10
> *Adresse: *4 allée des boulards
>                    78430  Louveciennes

i had the issue . I use Fusion 360 and when i send the part to cura i have
the issue
*Jérôme Ollivon*

*Mail*: ***@***.***
*Téléphone:* +33 6 81 55 04 10
*Adresse: *4 allée des boulards
                   78430  Louveciennes

MariMakes, 2022-11-10:
Hey @JOLLIVON,

Thanks for getting back to us 👍
How are you sending your model from Fusion 360 to Cura? 

Are you exporting your model to an STL?
[Are you Sending to Cura? ](https://knowledge.autodesk.com/support/fusion-360/learn-explore/caas/sfdcarticles/sfdcarticles/How-to-3D-print-a-stl-model-in-Fusion-360-to-a-Ultimaker-Cura-printer.html) 
Or are you using the [Fusion 360 integration](https://marketplace.ultimaker.com/app/cura/integrations/Autodesk/Fusion360)?

JOLLIVON, 2022-11-10:
Hello, I use the fusion360 integration with a specific application because
I use Creality slicer and Ultimate Cura.
*Jérôme Ollivon*

*Mail*: ***@***.***
*Téléphone:* +33 6 81 55 04 10
*Adresse: *4 allée des boulards
                   78430  Louveciennes

MariMakes, 2022-11-30:
Apologies for any inconvenience caused by our delayed response. 
To me, it doesn't seem like a bug in Ultimaker Cura, since it only seems to happen with a Fusion360 integration. 

Would you still like to keep this issue open or do you agree that we can close it?

closed 2022-12-14 by github-actions[bot]