Ultimaker/Cura #13777: Cura won't open gcode files from explorer

FerdinandoPH, 2022-11-09:
### Application Version

5.2.0

### Platform

Windows 11

### Printer

Anet E5

### Reproduction steps

1. I right-clicked a .gcode file
2. I clicked open with>search a program on my pc
3. I selected C:\Program Files\Ultimaker Cura 5.2.0\Ultimaker-Cura.exe

### Actual results

I get a popup saying: "the program cannot open that file extension"

### Expected results

I expected Cura to open and show me the GCODE file (as it used to do in 5.0 and below)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9973779/cura.log)

FerdinandoPH, 2022-11-09:
My bad, I meant 5.2.1

Anyways, all I get is this:
[image]

As you can see, the only option is "cura.exe" (which is from the old 4 version I uninstalled I believe), and it doesn't work.

And this is what happens when I manually select Ultimaker-Cura.exe

[image]

It says "the program you have selected cannot be associated with this filetype. Choose another program"

GregValiant, 2022-11-09:
Thanks for getting back.
This sounds like a Windows problem.  I didn't have any trouble with it in Win10 and that was opening a file with different versions of Cura.
Do you have the Gcode filename extension associated with an application?  If you right click on a gcode file and select "Properties" then one of them will be "Opens With".  I have mine set to a text editor (Notepad++) but using regular Windows Notepad will work.
Once Windows knows that it may (might???) allow you to open the file with Cura.
There have been a few problems reported by Windows 11 users that were never a problem with Win10.  This might be another one.  When you tell Windows to open a file with a particular piece of software it should open it.
I can't remember but when Cura is installed is there an option to associate certain file name extensions with it?  The developers will know.

Asterchades, 2022-11-10:
I'm not able to reproduce this under Win11 Pro x64 (22H2). It isn't in the default list of valid applications for the extension (seems a strange oversight of the installer), but manually selecting the Ultimaker-Cura.exe under Program Files\Ultimaker Cura 5.2.1 loads it into a new Cura instance, and places Ultimaker-Cura.exe into the list for future use (which also works).

On the off chance it was relevant I even renamed my gcode file to match yours and placed it in the same relative location (Documents\3dobj\3dprint), and it still worked. However I'm not using the same absolute file location as you (my username would likely be different) and my system language is English (AU), which may or may not be relevant.