Ultimaker/Cura #13779: Cura Bridging - Fan Inertia Variable

ablarry91, 2022-11-09:
### Is your feature request related to a problem?

When performing a bridging function on something like a temperature tower, the very first bridging element tends to sag due to too much thermal mass being extruded and/or insufficient cooling. During this very first pass, the blower fan is accelerating from a low RPM to full RPM and takes measurable time to reach its proper speed. With thermal mass extrusion being held constant, my problem is that cooling is not at set point the moment a bridge starts.

### Describe the solution you'd like

I would like a variable that effectively turns on the fan moments before the bridge is actually performed. This variable could be on the basis of acceleration, or more simply, the variable could be a dumb timer ("turn fan on 0.25s before bridging").

### Describe alternatives you've considered

I have considered:

- installing higher power blower, at the burden of turning this into a straight hardware problem
- reducing extrusion rate, at the risk of underextrusion symptoms
- reducing toolhead speed while bridging, allowing the fan to reach its top speed earlier into the bridge

### Affected users and/or printers

It will affect all users who are familiar with the bridging function and are trying to avoid the use of supports.

smartavionics, 2022-11-09:
See https://github.com/smartavionics/Cura/releases for a Cura variant that has had a Cooling Fan Lag setting since 2019.

MariMakes, 2022-11-10:
Hey @ablarry91,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13015
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13015

closed 2022-11-10 by MariMakes