Ultimaker/Cura #13780: Outside-In Wall Ordering on First Layer

ablarry91, 2022-11-09:
### Is your feature request related to a problem?

When printing fine details on the first layer, such as text on the underside of a benchy, or a PCB template, an inside-out wall ordering will tend to cause small "blob" walls to print before more substantial sections are put down. Printing outside-in would help, however, printing outside-in generally performs worse on subsequent layers in instances of unsupported overhangs. My problem is that the function that gives best first layer qualities worsens quality for subsequent layers.

### Describe the solution you'd like

I would like the `Wall Ordering` variable to have a boolean that says something to the effect of "First Layer Wall Ordering". If the boolean is selected, I have the option to specify a different wall ordering for layer 1.

### Describe alternatives you've considered

I have considered and/or have done:

- Reducing layer 1 print speeds, to a point where the first layer of a coffee cup footprint may take over an hour, to some success but causing ridiculous increases in print time
- Lowering viscosity of applied plastic by elevating first layer hotend/bed temperatures, to very little success
- Exploring alternative bed adhesion technologies, at the risk of turning this into a costly hardware problem
- Slicing the part in each wall ordering configuration, and manually cut/pasting layer 1 code, which works, but is tedious and error prone

### Affected users and/or printers

This will affect users who struggle with bed adhesion of fine features. This feature won't really matter for instances where large areas of plastic are in contact with the bed.

MariMakes, 2022-11-10:
Hey @ablarry91,

Thanks for your report 👍

When reading your request it reminds me of the Bottom Pattern Initial Layer setting.
I've seen quite some people changing this setting to Concentric to approach the behavior that you are describing.

I'll bring it up to our Print Profile Experts to see if this is something that they are considering improving. 
Fingers crossed 🤞

Fribb, 2022-12-21:
> @MariMakes I was not aware of that feature and will check it out! Thank you for your support!

As an alternative, you can change the wall ordering just for the first layer in a different way:

1. you place a support blocker
2. resize the support blocker to cover the whole print bed (or just the whole model) in X and Y and set the Z to the initial layer height. The following steps would then only apply to the first layer.
3. reset the position to 0,0,0 so it is flat on the bed
4. make sure you only select the large support blocker
5. above the "support blocker" option is the "Per model Settings" option
6. select "Modify settings for overlaps"
7. select "cutting mesh" in the drop-down menu
8. add the Wall ordering option and select "outside to inside" 

now, when you slice it, you will see that it will actually print from the outside to the inside and only for the first layer.

I agree, that this is annoying because you would always have to add that support blocker with each new project but at least that is a workaround

BarsMonster, 2023-01-27:
"Concentric" on first layer is currently not a feasible workaround, as Cura 5.2.1 prints concentric first layer in random order i.e. making adhesion problem even worse.

BarsMonster, 2023-01-27:
Relevant concentric bug is: https://github.com/Ultimaker/Cura/issues/10957
I.e. currently no non-manual workaround.

MariMakes, 2023-01-27:
Hey @BarsMonster,

We are aware that the ordering isn't correctly sequential in 5.2
We've put introduced a significant amount of improvements for the next 5.3 Cura release regarding ordering.
You can read more about them here: https://github.com/Ultimaker/CuraEngine/pull/1761 and https://github.com/Ultimaker/CuraEngine/pull/1794

BarsMonster, 2024-01-27:
Yes, this feature is still relevant.
I many cases I would also prefer printing out->in on first layer (for example right after brim), and in->out on subsequent.