Ultimaker/Cura #13781: Slcing error

OKCJim, 2022-11-09:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

Tried to slice a file.

### Actual results

Got an error

### Expected results

File should have sliced.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-11-10:
Without a project file and Cura log file we can't do an investigation.
If you would create a project file ("File | Save Project") and get the log file ("Help | Show Configuration Folder") and zip them up and post the Zip folder here then someone can take a look.

OKCJim, 2022-11-10:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/9985172/cura.log.zip)

Not sure what is causing this. I recently upgraded computers. This file sliced fine before.

OKCJim, 2022-11-10:
I didn't mean to close the issue early. This is my first time reporting a bug. Thank you for your patience.

OKCJim, 2022-11-11:
Also, if it helps, my new PC was running 5.2.1 and wouldn't slice this file. My old PC was running 4.13.1 and it will slice this file. I double checked and sliced it again on the old computer with 4.13.1. It still slices. I then downloaded 4.13.1 on the new PC and it too will slice this file. Hope that might be of some help? I would think that would point to something different in the settings?

Thanks

OKCJim, 2022-11-16:
Anything?

OKCJim, 2022-11-17:
I was able to slice it on 5.2.1 with the layer height set at 0.2 and Adaptive Layers turned off.  With thinner layers or Adaptive Layers turned on I get an error. With 4.13.1 I can slice with .12 layers and Adaptive Layers turned on. Does that point to anything in particular causing it?

Thanks for your help, and a big thanks for figuring out what could work. I was locked in on trying to replicate my old settings and didn't even think to experiment with different settings.

Jim

GregValiant, 2022-11-17:
When I opened your project file the Layer Height and Initial Layer Height were 0.5.  I thought that a bit much for a .4 nozzle and doubly so for such a detailed model.
Good that you got it figured out.

As far as pointing to things, I don't know.  Adaptive Layers is still in the "Experimental" section which kind of points to it still having issues.  Low layer heights and a wide range for adaptive to work can cause layer heights that the printer can't adjust to.  (Not enough resolution in the Z.)  To Cura they are just numbers but there could be a machine setting telling it that a .01 layer height is impossible for the printer.  Again, I don't really know so I'm surmising (taking a WAG.)

I'll leave this open for @MariMakes to check on when she gets back.

OKCJim, 2022-11-17:
Thanks for your help. I had completely missed that my initial layer height was 0.5. Trying to get the new PC set up and I must have over looked that.