Ultimaker/Cura #13782: Crash

TheFeralEngineer, 2022-11-10:
### Application Version

5.2.1

### Platform

windows 11

### Printer

elegoo neptune 3

### Reproduction steps

here's the crash log. It's pretty much at random with any open file. It seems to happen after a slice.

### Actual results

crash

### Expected results

no crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Faulting application name: Ultimaker-Cura.exe, version: 0.0.0.0, time stamp: 0x63500e7e
Faulting module name: igxelpicd64.dll, version: 31.0.101.3358, time stamp: 0x62ec9ec0
Exception code: 0xc0000005
Fault offset: 0x000000000022416e
Faulting process id: 0x0x5D4
Faulting application start time: 0x0x1D8F4A824B59420
Faulting application path: C:\Program Files\Ultimaker Cura 5.2.1\Ultimaker-Cura.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_f60bb83011df0c7e\igxelpicd64.dll
Report Id: 08f443f2-1834-48d3-8f53-4cf4a59dc2dd
Faulting package full name: 
Faulting package-relative application ID:

MariMakes, 2022-11-10:
Hey @TheFeralEngineer,

Thanks for your report 👍
Sorry to hear you are still encountering these crashes. 

[Do you have a custom printer loaded in Cura.](https://community.ultimaker.com/topic/41255-neptune-3-settings-for-cura/)
Or are you using [the Elegoo Cura ](https://www.elegoo.com/pages/3d-printing-user-support) to print on your Neptune 3? 

It might help more if you could share your configuration folder with logs included. 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2022-11-24 by github-actions[bot]

Gnitry, 2023-11-04:
I confirm that I have the same problem very often. On two different computers. After this bug happens everything is freezing for a few seconds (seems like Explorer is restarting also). And then some applications (if opened) that use video card stop working - for example Fusion 360 stopped rendering the model - need to be restarted.

This bug occurs when I am moving up/down the vertical slider to see my sliced model layers. Sometimes the bug occurs even without sliding.

This bug also was reproduced when running in PyCharm using latest sources. Logs and configs are inside archive.

[Uploading cura_bug.tar.gz…]()