Ultimaker/Cura #13783: slicing failed

3dprintrecycling, 2022-11-10:
### Application Version

5

### Platform

windows 10

### Printer

creality ender 5 pro

### Reproduction steps

adjusted retraction settings
rotated model 180 degrees
attempted to slice

### Actual results

slicing failed due to an unexpected error

### Expected results

model should have been sliced

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[project.zip](https://github.com/Ultimaker/Cura/files/9977196/project.zip)

GregValiant, 2022-11-10:
Thanks for the report.
I'm unable to duplicate this on my Win10 laptop and using your project file.

There is an oft-reported bug that seems to be triggered by the orientation of the model on the build plate.  You can try rotating the model again and see if there is a rotation angle that will fool the bug.

I wouldn't think the retraction distance has anything to do with this but seemingly unrelated settings can cause problems.
I'll mark this as a duplicate and the Cura team will take a look.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker