Ultimaker/Cura #13784: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses

leelufei, 2022-11-10:
### Application Version

5.3.1-alpha

### Platform

windows 10

### Printer

ultimaker s5

### Reproduction steps

1) successfully go through the instructions of https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source
2) .\venv\Scripts\pyinstaller.exe   .\cura_app.py
3)  PS D:\cura5.1.0\Cura> .\dist\cura_app\cura_app.exe
PyInstaller\loader\pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses

PS: I know that i have ignored some steps. Could you have some instructions about  how to generate cura install exe file on PC.

### Actual results

cura failed to start up

### Expected results

cura  starts up

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2022-11-10:
The message about the imp module deprecation is just a warning, and by itself that warning does not crash Cura.
You are going to need the information in the detailed crash report to see why Cura is crashing.

leelufei, 2022-11-11:
Aldo,  thanks for your attention.
According to the log, the issue looks like that cura_app.exe failed to get UM.
I used the pyinstaller to generate exe file. I must have missed something. 
Do  you have any instructions about how to generate cura exe file on Windows.

Here is the error log.
Traceback (most recent call last):
  File "cura_app.py", line 239, in <module>
  File "cura\CuraApplication.py", line 824, in run
  File "UM\Qt\QtApplication.py", line 381, in run
  File "UM\Application.py", line 268, in run
  File "cura\CuraApplication.py", line 493, in startSplashWindowPhase
  File "UM\Qt\QtApplication.py", line 235, in startSplashWindowPhase
  File "cura\CuraApplication.py", line 805, in _loadPlugins
  File "UM\PluginRegistry.py", line 382, in loadPlugins
  File "UM\PluginRegistry.py", line 436, in loadPlugin
UM.PluginError.PluginNotFoundError: Could not find plugin SentryLogger

MariMakes, 2022-11-11:
Hey @leelufei,

Welcome to the Ultimaker Cura Github 🚀 

We are aware of the issue, it's related to Python 3.12. 
We are currently running on Python 3.10 where this is not an issue. 

We've added a ticket to the backlog with the intent to improve this once we start upgrading Python. 

Thanks for the report! 👍

fieldOfView, 2022-11-11:
@leelufei, I would rather see the entire log (cura.log in %appdata%\cura\devel\), because that would likely show *why* the SentryLogger is not found. 

@MariMakes, the message about Python 3.12 is actually irrelevant here. AFAIK python 3.12 does not exist yet, outside of nightly developments builds. I am fairly confident that they are already running Python 3.10. The crash is something else going wrong.

wasmann, 2023-02-10:
Traceback (most recent call last):
  File "cura_app.py", line 239, in <module>
  File "cura\CuraApplication.py", line 824, in run
  File "UM\Qt\QtApplication.py", line 381, in run
  File "UM\Application.py", line 268, in run
  File "cura\CuraApplication.py", line 493, in startSplashWindowPhase
  File "UM\Qt\QtApplication.py", line 235, in startSplashWindowPhase
  File "cura\CuraApplication.py", line 805, in _loadPlugins
  File "UM\PluginRegistry.py", line 382, in loadPlugins
  File "UM\PluginRegistry.py", line 436, in loadPlugin
UM.PluginError.PluginNotFoundError: Could not find plugin SentryLogger

Thread 0x00002a60 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000da0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003128 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000028c0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003c7c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003d4c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004208 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000023bc (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x000037a4 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 177 in exceptHook

wasmann, 2023-02-20:
@fieldOfView @MariMakes

fieldOfView, 2023-02-20:
> I would rather see the entire log (cura.log in %appdata%\cura\devel), because that would likely show why the SentryLogger is not found.