Ultimaker/Cura #13787: Error in cura with stl (tinkercad)

eaba91, 2022-11-10:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Artillery SIDEWINDER X2

### Reproduction steps

1. Base for a pen, with shield of link (BOT) in tinkercad
2. after create export to cura 5.2.1
3. segment the stl file
4. 

### Actual results

nothing happens

### Expected results

nothing happens, 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-11-10:
Thanks for the report.
TinkerCad is notorious for creating models with surfaces flipped.  The slicer can't tell what is the outside of the model ad what is the inside and so slicing fails.
If you set the view to X-Ray Cura may provide you with a view of any error surfaces.
Often the models can be repaired in an app like MS 3D Builder (bundled with Windows) or you can use a program like Mesh Mixer.
That's one thing that might be happening.  Unless you provide us with a Project File ("File | Save Project") it's all guesswork from here.

eaba91, 2022-11-11:
Muchas gracias por su ayuda!

saludos.

fieldOfView, 2022-11-11:
> TinkerCad is notorious for creating models with surfaces flipped.

Is it though? TinkerCAD is not Sketchup. AFAIK, TinkerCAD is a solid modeler which produces fairly good models for 3d printing, whereas Sketchup is a surface modeler which was never intended for 3d printing.

GregValiant, 2022-11-11:
Nice catch.  That was my bad.  I was thinking SketchUp.
Still and all, checking the models for errors can't hurt.