Ultimaker/Cura #13789: CUra running extremely slow

j-snijder, 2022-11-11:
### Application Version

5.1.0 and 5.2.0 beta and 5.2.1

### Platform

Windows 11

### Printer

Ender 3pro

### Reproduction steps

Just slicing with standard settings.
0.16 or 0.2 layer
temp 215 / 70 for PLA
using skirt or brim doesn't change
retraction on or of doesn't change

 

### Actual results

The program is loading slow
The slicing takes very long
Saving takes very long

Everything is as slow as a snail on a frozen lake, not funny :(

### Expected results

The expected result should be normal loading times, normal slicing times and normal saving times

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

creating log file ? or saving the project file ? takes so long and then cura stops and freezes

j-snijder, 2022-11-11:
I have just deleted the auto orientation and auto start slicing plugin.

Now loading goes faster (as expected) and slicing and saving goes almost to normal.

If I have time I'll remove and test some more plugins that can (or at least are expecting to) slow thing down.

nallath, 2022-11-11:
It makes sense that the auto orientation plugin makes things slower, as it's doing quite a bit of calculation to determine the best orientation. 

It shouldn't influence the saving in any way though. The same is true for the autoslicing.

j-snijder, 2022-11-11:
Hi,Sorry to say that it did not help for long.After slicing one item, and removing it from the bed with ctrl+d cura is slow again.Something is very wrong with cura as of v 5.1.0, v4.8.1 (used by creality slicer does not have this problem.Also a v before v5, which I used to use, does not have this problem.Where can I download older versions to test ?This problem has to be solved 😡Regards, J. Snijder

MariMakes, 2022-11-30:
Hey @j-snijder,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 
Are you still stuck? 

I've seen similar reports before, but I'm having some trouble pinpointing the problem.
I'll need a little more information to help you troubleshoot. 
Can you share your configuration folder with us? 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

j-snijder, 2022-11-30:
Hi,

Well its inevitable that the bug was introduced around v5.0.0, because that version ran without any problems.\ at my system at least.

The reports are many and many more, the problem is not with Windows 11 in my case becaus I ran v 5.0.0 perfectly on Windows 11 and after that I have done no updating at windows.

My problems started with version 5.2.1 and before that I had no noticable problems.

Maybe its a smart idea to revoke every addon after that version and check where it starts.

In my opinion it has something to do with Cura searching for something.

What I did notice, as I emptied the temp folders in as well windows as the users maps that Cura starts faster and with its first object thats loaded after this there is less slower actions, but after one or two changes of settings it starts again and slows down massively.

I can not send you the files because Cura is absolutely irresponsive at the moment and I can not open the help section :)☹

Hopefully this helped a bit in determining and locating the problem.

Kind regards, Jan

Verzonden vanuit Mail voor Windows

Van: MariMakes
Verzonden: woensdag 30 november 2022 11:32
Aan: Ultimaker/Cura
CC: j-snijder; Mention
Onderwerp: Re: [Ultimaker/Cura] CUra running extremely slow (Issue #13789)

Hey @j-snijder,
Welcome to the Ultimaker Cura Github 🚀
Sorry, it took us a while to get back to you 😞
Are you still stuck?
I've seen similar reports before, but I'm having some trouble pinpointing the problem.
I'll need a little more information to help you troubleshoot.
Can you share your configuration folder with us?
You can share your configuration folder by:
1. Go to Help at the top bar of Cura
2. Select Show Configuration Folder
3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.
—

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

j-snijder, 2022-11-30:
I deleted some older cura stuff from my hdd and am in testing mode now, trying to find what cura is looking for, must be something like that imho

FOund it, please rename .zip into .7z as I don't have standard zipper

[5.2.zip](https://github.com/Ultimaker/Cura/files/10124038/5.2.zip)

MariMakes, 2022-11-30:
Hey @j-snijder,

Thanks for sharing. 
This is indeed what I was looking for.

Now I can have the same printers, materials and plug-ins as you. 
I rarely run my Cura in Dutch so I found some interesting language errors. 
[image]

Unfortunetly my Cura is running as expected, so I can't recreate your issue. 
I'll have someone in the team take a look at your logs, maybe they can see something I've overlooked. 
Fingers crossed that they can find something.

j-snijder, 2022-11-30:
I think I found (at least part of the problem by now :) )

What I did is locating all the instances of previous cura installations and leftovers like temp  filled with every kind of whatever :)

I found 3 maps that have some (big) effect, knowing the cura map in program files, and the maps in appdata roaming.

I deleted everything in the temp maps I found there and restarted Cura, and BOOM! it ran like before, without any noticably problem.

So I was right, Cura is looking for old stuff in these maps and gets kinda confused, resulting a dicicesless action on the end of cura.

It looks like when updating cura stuff is left behind that causes that slowing down.

I'll test some more in the upcoming hours and days to see what happens.

j-snijder, 2022-11-30:
PS, as it occured right after updating to cura 5.2.1 I suspect cura of not neatly cleaning up the temp or cache maps leaving some stuff behind that disorientates cura.

I'll try to test that as I go along using cura the way I am used to.

j-snijder, 2022-11-30:
As for the translation there some other thing I noticed.

Linewidth is translated in lijnbreedte which, correct me if I am wrong, is actually not right.

I found that line width is ment to be the distance between the printed lines, is it ?
If so then its not lijnbreedte you are talking about but the space between printed lines which in correct Dutch is lijnafstand :)

Where a lower number puts the lines closer together and a bigger number widens up the space between them, or am I not understanding the concept ?

MariMakes, 2022-11-30:
Hey @j-snijder,

Thanks for the extra research. 🎉 
Are you unstuck, now?
What do you think, can we close theisissue? 

It seems to be related to: https://github.com/Ultimaker/Cura/issues/13446
I'll make that it's included in the implementation.

j-snijder, 2022-11-30:
Please leave it open for a short while so I can report back through this post as I am testing along.

If after a week there is nothing to report you can close it :)

j-snijder, 2022-11-30:
Hi, UPDATE:

The problem is nót related to the item you mentioned.

As I am in the middle of a big thing with many parts I am currently slicing and notice that with every part done CUra slows down, so I guess the problem is with Cura not cleaning up the cache after a file is done, saved to disk, and cleared the bed with CTR+D

I am going to check if this happens also when closing down Cura after every slice but I expect no difference as Cura should clean its cache as supposed after removing the part.

Regards, Jan.

I'll keep you updated.

j-snijder, 2022-12-01:
As I am slicing along I come to the believe that the problem is not only Windows 11, unless I am mistaken, to me it looks clearly a matter of Cura not cleaning up his mess, or whatever causes that.

As I cleanup the caches in appdata myself Cura keeps running well on my Windows 11 machine, and man do I have that one in a stressed state :) I like it to open all programs I use together when working with my 3D printer so thats how I am testing this at the moment op typing.
IF it would be a pure Windows 11 problem I;ll eat my sneakers :)

j-snijder, 2022-12-01:
What did I do now ? After fully freezing I deinstalled all cura, really all and let Revo remove everything.
Then reinstalled v 5.1.0, which I knew was working fine.
I discovered I forgot installing v 5.1.1 so I did, BAMMMM
Now the problems started again and cura freezes at adding custom filament.

Could it be that the newly added printers (Ultimaker) are causing the problems ?

Please investigate, I hope I accidentally discovered the problem.

j-snijder, 2022-12-01:
Removed some old filaments and decoupled some that were linked together.

Except for the color changer I notice no strange slowdowns.

Monday I'll go deeper into this, sorry, weekend with my grandchildrens :)